mne-tools/mne-python #8583: More control over plot settings

eort, 2020-11-30:
#### Describe the new feature or enhancement
Would it be possible to give users more control over (some of the) plotting parameters? 
in my case, I wanted to plot multiple ERPs in a single figure. WIth `plot_evoked`, I could extract all the axes and gather all the plots in a single figure, however, the layout was not really nice (a lot of whitespace). I wished to constrain the layout, but couldn't do so because in `plot_evoked` `set_tight_layout` is hardcoded to `True`, which is incompatible with a constrained layout.

#### Describe your proposed implementation
If I understand the structure correctly, there are front-end auxiliary plotting functions that users interact with and the actual plotting functions that are called in the background and do the plotting. As it seems, only a subset of the arguments that the plotting functions have are being used by those auxiliary functions. Would it be possible to pass those "hidden" arguments as `kwargs`? Like that the API would not be cluttered with secondary plotting parameters, but still allow for more flexibility? 

#### Additional comments
As mentioned before, I only had a look at the `plot_evoked` function. I don't know in how far something similar is also relevant for other plotting functions.

agramfort, 2020-11-30:
you tried to do the right thing

set_tight_layout

should probably not be used systematically.

regarding kwargs I prefer to avoid them as much as possible as
they are almost impossible to document and can lead to silent bugs

can you open a PR with the changes that are necessary to address your
needs and looking at the diff we see how to move forward?

>

drammock, 2020-11-30:
@eort I'm in the process of refactoring a lot of the 2D plotting functions (see #7751). One of the things I am doing is eliminating hard-coded calls to `tight_layout` whenever possible. Please @-mention me when you open a PR, I'd be happy to chime in or help out...  `plot_evoked` was actually next on my TODO list since it shares a lot of code with the `plot_psd` functions (which I just finished last week).

eort, 2020-11-30:
Sounds very nice, thanks!

> since it shares a lot of code with the plot_psd functions (which I just finished last week).

cool, I was having similar issues with `plot_psds` (it doesn't have a title keyword).

> Please @-mention me  when you open a PR
will do

drammock, 2020-11-30:
>  I was having similar issues with `plot_psds` (it doesn't have a title keyword).

We definitely want to support that use case one way or another, but I agree with @agramfort about allowing users to pass matplotlib kwargs through the MNE-Python plotting functions --- handling them correctly can dramatically complicate our code and makes it easier for silent bugs to accumulate.  What I have been trying to do is make all the 2D plots amenable to the standard matplotlib adjustments *after* figure creation, so that things like `fig.subplots_adjust(top=0.8)` or `fig.suptitle('my custom title\nwith a newline in it')` will work as expected and not require extra steps or workarounds.

I am keen to hear your "laundry list" --- are there things other than adding titles that you are finding difficult to accomplish?

eort, 2020-12-02:
> I am keen to hear your "laundry list" --- are there things other than adding titles that you are finding difficult to accomplish?

So far, I haven't encountered much that I had trouble with, as I just started optimizing the visualization of my data. What might be relevant to you is: 

- inconsistent parameter naming for passing axes (sometimes `ax` as in `plot_raw_psd`, sometimes `axes` as in `plot_evoked`)
- the interactions with the plot were confusing whenever more than one axis is returned (encountered in `plot_evoked`). For example, even though I passed only two axes each on two separate calls to the function into the function, four axes got returned for each call. Other things were that adding titles/subtitles as well as x/y labels confused me e.g. labels disappeared). Eventually it looked like what I would have expected, but I don't know what caused the issue in first place and how it was resolved (probably mistakes on my end)

Here are also a  few other plot-related things that came to my mind, in a wider sense of your question:

- when plotting raw data (in `plot_raw`). I would like to group the channels by more than one variable (i.e. location and type). I haven't tried whether this could be done by combining the `group_by` and the `picks` arguments, though.  
- when plotting epochs, I would like to be able to merely mark epochs without instantly dropping them. So, storing indices of to-be-dropped epochs in the epoch instance and drop them later with an explicit dropping call. 
- it would be cool to have a [waterfall plot](https://www.mathworks.com/help/matlab/ref/waterfall.html) added to mne's plotting functions. That is basically a 3d version of `plot_image`. I have seen power spectra plotted in that way (over frequencies and trials) and was impressed how nicely noisy trials can be identified. But yeah, given that this would be a 3d-plot it probably doesn't fit your agenda. 

As I said, I haven't spend a lot of time exploring all the ways to amend mne plots. It is quite likely that I have just missed obvious solutions. I will be spending quite a bit of time in the next weeks with plotting. I will report here (or in [#7751](https://github.com/mne-tools/mne-python/issues/7751)) whenever I encounter something.

PR #8621 (2020-12-07): VIZ, MAINT: remove set_tight_layout from evoked plotting code
closes #8583 
closes #8597 (supersedes)

removes `set_tight_layout` from our internal evoked plotting code. This was a problem for users who were providing their own axes to `evoked.plot()`. Also tweaks the figure margins that we set when axes are not provided, to approximate what tight_layout would have yielded.

This is a short-term fix; soon evoked plots will use our new figure classes anyway.

cc @eort
closed 2020-12-07 by larsoner