mne-tools/mne-python #8587: BUG, ENH, DOC:  Overwriting of raw with certain tools in simulation/raw.py

NeuroLaunch, 2020-11-30:
#### Describe the bug
In `add_ecg()`, `add_eog()`, and `add_chpi()` functions in mne/simulation/raw.py, the newly added waveforms overwrite the existing raw data (e.g. as created with `simulate_raw()` and `add_noise()`). 

Aside from this bug, I would suggest enhancements related to dealing with the _raw.first_samp_ parameter and its behavior with event and head position information when creating simulated data.

#### Steps to reproduce
The following is an edited snippet from my own code:
```
raw_sim = simulate_raw(raw.info, source_simulator, src=src, bem=bem, head_pos=head_pos)
add_noise(raw_sim, cov=cov)    # this part is ok

add_chpi(raw_sim, head_pos=head_pos)    # this part erases the above data in 'raw_sim'
add_ecg(raw_sim)    # so does this
```

Additionally, the bug is evident in the simulation example at https://mne.tools/stable/auto_examples/simulation/plot_simulated_raw_data_using_subject_anatomy.html, and at least one other.

#### Expected and actual results
In the MNE documentation example linked above, the source activity and structured noise created in _raw_sim_ are missing after the EOG and ECG steps.  As a result, the brain map source estimations reflect only the ECG artifact.

#### Additional information
I have identified the source of the bug described and plan to make my own PR with a simple fix.  However, I thought it would be a good opportunity to also make documentation revisions and code enhancements related to simulations, such as a fuller description of the output in the examples and inclusion of the _raw.first_samp_ information when adding events to the SourceSimulator object used in `simulate_raw()`.

agramfort, 2020-11-30:
oh boy...

thanks for catching that !

can you start a PR to fix this?

thanks

>

NeuroLaunch, 2020-12-01:
@agramfort  Yes, I can do an immediate fix for the bug soon.  Is it OK if I commit related changes (for the documentation and  enhancements I mentioned) to the same PR as I get to them, or is it preferable that I split the additional work into a second PR?

agramfort, 2020-12-01:
if you prefer to do one PR at a time it's good.

it will be merged faster this way

PR #8633 (2020-12-16): FIX: prevent raw._data overwrite from _add_exg and _add_chpi
#### Reference issue
Fixes #8587 

#### What does this implement/fix?
in mne/simulation/raw.py  - functions _add_exg and _add_chpi, the code set the raw._data to a variable "data".
The variable "data" was written over by the output of these functions.  This would leave the raw._data as the last operation performed. The fix is to sum the raw._data numpy array with the output from the functions.  (+= vs =).
closed 2020-12-16 by larsoner