mne-tools/mne-python #8593: Adding SEEG electrodes's info in 3D

BarryLiu-97, 2020-12-02:
Hi, recently I've been thinking that showing some info about the SEEG electrodes may be very useful when you move your mouse to the electrode in the figure produced by plot_alignment, for example, the name, the MNI coordinates, and which brain area the electrode is in. 
**Suggestion:**
Adding some info of SEEG electrodes to the figure produced by plot_alignment.
For now, I think these may be useful:
(1) the channel name
(2) the MNI coordinates
(3) the brain area that the electrodes belong to

Now I have figured out how to distinguish the brain area of each electrode belongs, but I don't know how to add the info to the 3D figure when moving the mouse to the electrode.
[image]
This is a sample of the info on the brain area.
[image]

BarryLiu-97, 2020-12-02:
May be like this
 
[image]

BarryLiu-97, 2020-12-02:
[zone_test.zip](https://github.com/mne-tools/mne-python/files/5626828/zone_test.zip)

agramfort, 2020-12-02:
can you show an example maybe from another software?

how would you interact with it?

would you use colors? show a text in 3D?

>

BarryLiu-97, 2020-12-02:
> can you show an example maybe from another software? how would you interact with it? would you use colors? show a text in 3D?
Sorry, I just heard from the Pi in my lab, but brainstorm should have this kind of function.
Here is the detail: When you move your mouse to a specific electrode and click on it, the circle that represents the electrode will be highlighted, and in a sub-window, there will be some detailed information of it.

agramfort, 2020-12-02:
hum... not so easy to do... but I see what you mean

>

BarryLiu-97, 2020-12-03:
> hum... not so easy to do... but I see what you mean
> 
Yes

BarryLiu-97, 2020-12-05:
> hum... not so easy to do... but I see what you mean
> 
While the key to implementing this function is  recognizing which electrode the mouse is pointing, then we just extract the info from the table we calculate and show them on a sub-window

BarryLiu-97, 2020-12-05:
So I think if we can show the electrodes' name in this 3D viewer, then what we need to do is simple.

agramfort, 2020-12-05:
do you think you can give it a try?

Unfortunately I don't have funded projects using sEEG data now to be able
to support this.

>

BarryLiu-97, 2020-12-05:
> do you think you can give it a try? Unfortunately I don't have funded projects using sEEG data now to be able to support this.
> 
I was trying. I didn't find any API or demo that showed that pyvista(or pyvistaqt) could support the move_mouse,  while I have some sample seeg data and its MNI file and the script that decodes where the electrode is. If you could give me some advice like a basic protocol would be great.

larsoner, 2020-12-08:
I think we can get this almost for free once we allow showing sensors / sensor data in _Brain, so I'll close this for #8382

closed 2020-12-08 by larsoner