mne-tools/mne-python #8599: BUG: Bug with `pip pre` / sklearn

larsoner, 2020-12-03:
@agramfort maybe a quick one for you:
```
____________________ test_get_coef_multiclass_full[4-10-2] ____________________
mne\decoding\tests\test_base.py:302: in test_get_coef_multiclass_full
    scores = cross_val_multiscore(time_gen, X, y, cv=cv, verbose=True)
mne\decoding\base.py:441: in cross_val_multiscore
    scores = parallel(p_func(clone(estimator), X, y, scorer, train, test,
mne\decoding\base.py:441: in <genexpr>
    scores = parallel(p_func(clone(estimator), X, y, scorer, train, test,
mne\decoding\base.py:474: in _fit_and_score
    X_train, y_train = _safe_split(estimator, X, y, train)
c:\hostedtoolcache\windows\python\3.8.6\x64\lib\site-packages\sklearn\utils\metaestimators.py:198: in _safe_split
    if _is_pairwise(estimator):
c:\hostedtoolcache\windows\python\3.8.6\x64\lib\site-packages\sklearn\base.py:844: in _is_pairwise
    pairwise_tag = _safe_tags(estimator, key="pairwise")
c:\hostedtoolcache\windows\python\3.8.6\x64\lib\site-packages\sklearn\utils\_tags.py:62: in _safe_tags
    raise ValueError(
E   ValueError: The key pairwise is not defined in _get_tags() for the class GeneralizingEstimator.
```

closed 2020-12-03 by agramfort

probeadd, 2020-12-22:
I just got this error messages, what shoud I do to my scikit-learn?

drammock, 2020-12-22:
@probeadd this bug was fixed 19 days ago.  The current stable version (0.22) was released 5 days ago.  So if you update to version 0.22  (or the current master version)  it should go away.  If you're still encountering this bug after upgrading MNE-Python, please open a new issue rather than commenting again here.

rahul2227, 2021-03-08:
I have the updated version of the MNE but still, I am getting this error now. I am running the same code as I designed before and I ran it without any problems. But now the same code is giving me this error.
```
ValueError                                Traceback (most recent call last)
<ipython-input-49-98fec0085f41> in <module>
     41 
     42 
---> 43 scores = cross_val_score(clf, epochs_data_train, labels, cv=cv, n_jobs=1)
     44 # scores = cross_val_score(clf, epochs_data_train, labels, cv=5, n_jobs=1)
     45 

~\AppData\Local\Continuum\anaconda3\envs\mne\lib\site-packages\sklearn\utils\validation.py in inner_f(*args, **kwargs)
     61             extra_args = len(args) - len(all_args)
     62             if extra_args <= 0:
---> 63                 return f(*args, **kwargs)
     64 
     65             # extra_args > 0

~\AppData\Local\Continuum\anaconda3\envs\mne\lib\site-packages\sklearn\model_selection\_validation.py in cross_val_score(estimator, X, y, groups, scoring, cv, n_jobs, verbose, fit_params, pre_dispatch, error_score)
    438     scorer = check_scoring(estimator, scoring=scoring)
    439 
--> 440     cv_results = cross_validate(estimator=estimator, X=X, y=y, groups=groups,
    441                                 scoring={'score': scorer}, cv=cv,
    442                                 n_jobs=n_jobs, verbose=verbose,

~\AppData\Local\Continuum\anaconda3\envs\mne\lib\site-packages\sklearn\utils\validation.py in inner_f(*args, **kwargs)
     61             extra_args = len(args) - len(all_args)
     62             if extra_args <= 0:
---> 63                 return f(*args, **kwargs)
     64 
     65             # extra_args > 0

~\AppData\Local\Continuum\anaconda3\envs\mne\lib\site-packages\sklearn\model_selection\_validation.py in cross_validate(estimator, X, y, groups, scoring, cv, n_jobs, verbose, fit_params, pre_dispatch, return_train_score, return_estimator, error_score)
    244     parallel = Parallel(n_jobs=n_jobs, verbose=verbose,
    245                         pre_dispatch=pre_dispatch)
--> 246     results = parallel(
    247         delayed(_fit_and_score)(
    248             clone(estimator), X, y, scorers, train, test, verbose, None,

~\AppData\Local\Continuum\anaconda3\envs\mne\lib\site-packages\joblib\parallel.py in __call__(self, iterable)
   1039             # remaining jobs.
   1040             self._iterating = False
-> 1041             if self.dispatch_one_batch(iterator):
   1042                 self._iterating = self._original_iterator is not None
   1043 

~\AppData\Local\Continuum\anaconda3\envs\mne\lib\site-packages\joblib\parallel.py in dispatch_one_batch(self, iterator)
    857                 return False
    858             else:
--> 859                 self._dispatch(tasks)
    860                 return True
    861 

~\AppData\Local\Continuum\anaconda3\envs\mne\lib\site-packages\joblib\parallel.py in _dispatch(self, batch)
    775         with self._lock:
    776             job_idx = len(self._jobs)
--> 777             job = self._backend.apply_async(batch, callback=cb)
    778             # A job can complete so quickly than its callback is
    779             # called before we get here, causing self._jobs to

~\AppData\Local\Continuum\anaconda3\envs\mne\lib\site-packages\joblib\_parallel_backends.py in apply_async(self, func, callback)
    206     def apply_async(self, func, callback=None):
    207         """Schedule a func to be run"""
--> 208         result = ImmediateResult(func)
    209         if callback:
    210             callback(result)

~\AppData\Local\Continuum\anaconda3\envs\mne\lib\site-packages\joblib\_parallel_backends.py in __init__(self, batch)
    570         # Don't delay the application, to avoid keeping the input
    571         # arguments in memory
--> 572         self.results = batch()
[... 49 lines cut ...]

~\AppData\Local\Continuum\anaconda3\envs\mne\lib\site-packages\sklearn\base.py in _get_tags(self)
    329                 # but might do redundant work in estimators
    330                 # (i.e. calling more tags on BaseEstimator multiple times)
--> 331                 more_tags = base_class._more_tags(self)
    332                 collected_tags.update(more_tags)
    333         return collected_tags

~\AppData\Local\Continuum\anaconda3\envs\mne\lib\site-packages\sklearn\pipeline.py in _more_tags(self)
    628     def _more_tags(self):
    629         # check if first estimator expects pairwise input
--> 630         return {'pairwise': _safe_tags(self.steps[0][1], "pairwise")}
    631 
    632     # TODO: Remove in 1.1

~\AppData\Local\Continuum\anaconda3\envs\mne\lib\site-packages\sklearn\utils\_tags.py in _safe_tags(estimator, key)
     60     if key is not None:
     61         if key not in tags:
---> 62             raise ValueError(
     63                 f"The key {key} is not defined in {tags_provider} for the "
     64                 f"class {estimator.__class__.__name__}."

ValueError: The key pairwise is not defined in _get_tags() for the class CSP.
```
Previously the same code worked well for me.

agramfort, 2021-03-08:
can you merge the main branch into yours?

>

PR #8601 (2020-12-03): fix new sklearn tags
closes #8599