mne-tools/mne-python #8604: 'mne.stc_to_label' only generate the labels in left hemisphere

perevo, 2020-12-04:
#### Describe the bug
'mne.stc_to_label' only generate the labels in left hemisphere. The data in right hemisphere is neglected automatically.

#### Additional information
Platform:      Linux-5.4.0-56-generic-x86_64-with-glibc2.10
Python:        3.8.5 (default, Sep  4 2020, 07:30:14)  [GCC 7.3.0]
Executable:    /home/jialiu/anaconda3/envs/mne/bin/python
CPU:           x86_64: 12 cores
Memory:        125.7 GB

mne:           0.21.1
numpy:         1.19.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.2
matplotlib:    3.3.2 {backend=Qt5Agg}

sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.2.0
cupy:          Not found
pandas:        1.1.3
dipy:          1.3.0
mayavi:        4.7.2
pyvista:       0.27.2 {pyvistaqt=0.2.0, OpenGL 4.5.0 NVIDIA 450.80.02 via Quadro M4000/PCIe/SSE2}
vtk:           9.0.1
PyQt5:         5.15.1

agramfort, 2020-12-04:
can you provide a script so we can replicate easily the issue? thanks

>

perevo, 2020-12-04:
I figured out finally, the output should be separated for left and right hemispheres :
_left_labels, right_labels = mne.stc_to_label(...)_
The Doc and the Example of this function is a little of misleading.
Thanks for reply.

agramfort, 2020-12-04:
can you send a PR to improve the documentation?

>

perevo, 2020-12-04:
ok, I am a novice at github and python. I will try it.

drammock, 2020-12-04:
That's ok! Don't hesitate to ask if you have questions or get stuck.
-------- Original Message --------

larsoner, 2023-10-01:
Based on a quick look at our tests, I think things should be working okay. I'm going to close this but @perevo if you manage to come up with a minimal example from the `sample` dataset let's reopen and fix the bug!

closed 2023-10-01 by larsoner