mne-tools/mne-python #8607: Select good epochs instead of bad epochs in epoch.plot

fabriceduprat, 2020-12-04:
#### Describe the new feature or enhancement
When data is noisy its possible to get much more bad epochs than good epochs, thus it is very time consuming to click on all bad epochs in the epoch.plot figure.

#### Describe possible alternatives
1) Add a key to select all epochs as bad and then we just have to click on good ones to unselect them and set them as good
2) Add a key to "reverse" the selection to mark as bad all the non clicked epochs and vice-versa

#### Additional comments
I tried to reverse the content of epoch.drop_log after selecting bad epochs but there are many assertions on the lengths of various other arrays and I did not succeed ...
Thank you very much if it is possible, or if it already exists, but i did not find it in documentation and github issues.

mmagnuski, 2020-12-06:
Would shift-clicking to select a range of epochs work for you? This shouldn't be too hard to implement if shift key is free.

IIRC the epochs are dropped when you close the gui, so to work around your issue you would have to make a copy of the data and select epochs to retain on this copy. These epochs will be removed so you will be able get their indices from `.drop_log` of the copied epochs (I don't remember if its trivial - for a long time I was using my own small function for that). Once you have the indices you could perform a set difference (between full indeces and the removed ones) to get the reversed indices. Then you can use the reversed indices on the original epochs with `epochs.drop()`.

fabriceduprat, 2020-12-09:
Hi

 

Thank you very much for your suggestion, I will try that, I tried to do it “manually” but its true that epochs.drop must be much better !

 

Best regards

 

 

********************************************************************************************************
Dr Fabrice DUPRAT
Chargé de Recherche INSERM

Phone : +33 (0)4 93 95 77 53
Fax: +33 (0)4 93 95 77 08
Email: duprat@ipmc.cnrs.fr <mailto:duprat@ipmc.cnrs.fr> 
Personal web site: http://www.ipmc.cnrs.fr/~duprat/

 

Institut de Pharmacologie Moléculaire et Cellulaire (IPMC)
CNRS UMR 7275
660 Route des Lucioles
06560 Valbonne
FRANCE
********************************************************************************************************

larsoner, 2023-10-01:
Sounds like there is a usable solution here so I'll close

closed 2023-10-01 by larsoner