mne-tools/mne-python #8609: CI, MAINT: circle failing again

drammock, 2020-12-04:
Circle builds are failing on the MxNE example with this:

```
Traceback (most recent call last):
  File "/home/circleci/project/examples/inverse/plot_time_frequency_mixed_norm_inverse.py", line 126, in <module>
    brain = stc.plot('sample', 'inflated', 'rh', views='medial',
  File "/home/circleci/project/mne/source_estimate.py", line 653, in plot
    brain = plot_source_estimates(
  File "<decorator-gen-139>", line 24, in plot_source_estimates
  File "/home/circleci/project/mne/viz/_3d.py", line 1794, in plot_source_estimates
    return _plot_stc(
  File "/home/circleci/project/mne/viz/_3d.py", line 1984, in _plot_stc
    brain.setup_time_viewer(time_viewer=time_viewer,
  File "/home/circleci/project/mne/viz/_brain/_brain.py", line 416, in setup_time_viewer
    self._configure_sliders()
  File "/home/circleci/project/mne/viz/_brain/_brain.py", line 719, in _configure_sliders
    value=self._data['smoothing_steps'],
KeyError: 'smoothing_steps'
```

(see here: https://app.circleci.com/pipelines/github/mne-tools/mne-python/5139/workflows/937701e3-a227-414d-a747-7c65e415acbe/jobs/23925)

@GuillaumeFavelier knows what went wrong perhaps?

GuillaumeFavelier, 2020-12-04:
I'm on it!

GuillaumeFavelier, 2020-12-04:
It seems caused by https://github.com/mne-tools/mne-python/blob/5a62f4719f28733687b5dd1f8fb9674f9f641c2f/mne/viz/_3d.py#L1911-L1912

closed 2020-12-07 by larsoner