mne-tools/mne-python #8615: Source distance from centre

olafhauk, 2020-12-07:
#### Describe the new feature or enhancement
Following PR #8534 ("source depth"), I would find it useful to have a method for source spaces compute_distance_to_center() which returns each vertex' distance to an origin of vertex coordinates. This could be useful to evaluate source/head/forward models without specifying a particular sensor configuration.
A possible API could be:

```
def compute_distance_to_centre(src, origin=None):
    """Compute distances between vertices and their origin.

    Parameters
    ----------
    src : instance of SourceSpaces
        The object with vertex positions for which to compute distances to origin.
    origin: None | str
        The origin of vertex coordinates to use for distance computations. Can be
        * None (default): Compute Euclidean norm of vertex coordinates (i.e. origin [0, 0, 0].
        * Talairach: Compute Euclidean norm of vertex coordinates in Talairach coordinate system.
        * MNI: Compute Euclidean norm of vertex coordinates in Talairach coordinate system.
        * COG: Compute Euclidean norm of vertex coordinates with respect to their center of gravity.

    Returns
    -------
    distances : array of shape (n_vertices,)
        The distances (in metres) from origin to vertex locations.
    """
```
#### Describe your proposed implementation
Ideally as method for source space objects. 

#### Describe possible alternatives
Could also be a function in source_space.py (similar to compute_distance_to_sensors).

#### Additional comments
The different options would require transformations to MNI and Talairach coordinate systems. I would need help with that.

agramfort, 2020-12-07:
@olafhauk let's leave this open and see how to leverage this in the
future to help mne users in
their data analysis.

I am a bit worried to add some many functions to public API.

olafhauk, 2020-12-07:
Ok.

agramfort, 2020-12-07:
note I am not saying no. Just let's see in a few months if we still need it
and we can reconsider
In the meantime you can use such a function for your local analysis.

>

olafhauk, 2020-12-07:
Absolutely fine. But just as a little P.S.: Is it possible to add this as a non-public method/function, so I could already use/test it within MNE?

agramfort, 2020-12-07:
if it's just for you can you put it in a private project to start with?

>

drammock, 2020-12-07:
> Is it possible to add this as a non-public method/function, so I could already use/test it within MNE?

On your fork of MNE, you can do whatever you want :)   If locally you add the function to a branch "dist_from_centre" then you can run any tests you want to locally while on that branch (e.g. `pytest mne/viz/tests/test_raw.py` or whatever), and even build the documentation locally: there's a makefile in `/doc` folder, you can run `make html_dev-noplot` to generate the full website without actually executing any of the tutorial code, or run `PATTERN=regex_for_tutorial_filenames make html_dev-pattern` to fully build one or a few of the examples/tutorials too.  After it finishes, `make show` will pop open the local build of the website.

larsoner, 2020-12-07:
> if it's just for you can you put it in a private project to start with?

OTOH if you want it to make things like testing some functionality already in MNE (or something you're going to add) easier, it would make sense to have it be some private function somewhere.