mne-tools/mne-python #8617: MemoryError when apply inverse

1101AlexZab1011, 2020-12-07:
#### Description
I created single 60-seconds epoch and tried to apply inverse, but MemoryError rose.

#### Minimal working example
inv_file = './path/to/raw-inv.fif'
snr = 0.5
lambda2 = 1.0 / snr ** 2
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)
raw = rawcrop(tmax=120)
events = np.array([[
        raw.first_samp + raw.time_as_index(120/2)[0],
        0,
        1
    ]]) # one event at the middle of the data
epochs = mne.Epochs(raw, events, tmin=-30, tmax=30,
                        preload=True)
evoked = epochs.average()
inv = mne.minimum_norm.read_inverse_operator(inv_file) # reading of the inverse solution from its file
stc = mne.minimum_norm.apply_inverse(evoked, inv, lambda2, 'sLORETA')

#### Expected results
Computing sLORETA for 1 minute of data

#### Actual results

---------------------------------------------------------------------------

MemoryError                               Traceback (most recent call last)

<ipython-input-20-edc93a083f21> in <module>
      5 else:
      6    
----> 7     stc = mne.minimum_norm.apply_inverse(evoked, inv, lambda2, 'sLORETA')
      8     path = res_sLORETA_folder
      9 

<decorator-gen-324> in apply_inverse(evoked, inverse_operator, lambda2, method, pick_ori, prepared, label, method_params, return_residual, use_cps, verbose)

~/anaconda3/envs/NodesEstimation/lib/python3.8/site-packages/mne/minimum_norm/inverse.py in apply_inverse(evoked, inverse_operator, lambda2, method, pick_ori, prepared, label, method_params, return_residual, use_cps, verbose)
    894     .. footbibliography::
    895     """
--> 896     out = _apply_inverse(
    897         evoked, inverse_operator, lambda2, method, pick_ori, prepared, label,
    898         method_params, return_residual, use_cps)

~/anaconda3/envs/NodesEstimation/lib/python3.8/site-packages/mne/minimum_norm/inverse.py in _apply_inverse(***failed resolving arguments***)
    937     K, noise_norm, vertno, source_nn = _assemble_kernel(
    938         inv, label, method, pick_ori, use_cps=use_cps)
--> 939     sol = np.dot(K, evoked.data[sel])  # apply imaging kernel
    940     logger.info('    Computing residual...')
    941     # x̂(t) = G ĵ(t) = C ** 1/2 U Π w(t)

<__array_function__ internals> in dot(*args, **kwargs)

MemoryError: Unable to allocate 27.4 GiB for an array with shape (61374, 60001) and data type float64

#### Additional information

Platform:      Linux-5.4.0-56-generic-x86_64-with-glibc2.10
Python:        3.8.5 (default, Sep  4 2020, 07:30:14)  [GCC 7.3.0]
Executable:    /home/user/anaconda3/envs/NodesEstimation/bin/python
CPU:           x86_64: 8 cores
Memory:        11.7 GB

mne:           0.21.2
numpy:         1.19.4 {blas=openblas, lapack=openblas}
scipy:         1.5.4
matplotlib:    3.3.3 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.23.2
numba:         0.52.0
nibabel:       3.2.0
cupy:          Not found
pandas:        1.1.4
dipy:          1.3.0
mayavi:        Not found
pyvista:       0.27.3 {OpenGL 4.5.0 NVIDIA 450.80.02 via GeForce GTX 1050 Ti/PCIe/SSE2}
vtk:           9.0.1
PyQt5:         5.9.2
None

agramfort, 2020-12-07:
what is your sampling rate?

if you are at 1kHz and you have 10^3 vertices in the source space
it means you need to have in memory

60 * 1e^3 * 10^3 * 8 Bytes

that's about 10GB of RAM just for this. You have only 12GB of RAM on your
machine.

>

1101AlexZab1011, 2020-12-07:
Yes, it is 1kHz
So I just need a machine with a lot of RAM?

agramfort, 2020-12-07:
or change your logic in the computation or downsample your data.

>

1101AlexZab1011, 2020-12-07:
Thank you very much

closed 2020-12-07 by drammock