mne-tools/mne-python #8620: Source orientations in get_point_spread()/get_cross_talk()

olafhauk, 2020-12-07:
#### Describe the new feature or enhancement
get_point_spread()/get_cross_talk() can currently only deal with fixed orientations, even though the resolution matrix can be of dimension (n_orient_inv * n_dipoles, n_orient_fwd * n_dipoles). This is undesirable (one may be interested in PSFs/CTFs for different source orientations), and this is currently not dealt with properly. I would like to propose an additional option 'vector' for those functions in order to solve this.

#### Describe your proposed implementation
The new API could look a bit like this. It's a bit hairy to describe the different combinations of n_orientations for forward solution and inverse operator and their consequencies for PSFs and CTFs. The default options will hopefully be easy to interpret.

```
def _get_psf_ctf(resmat, src, idx, func, mode, n_comp, norm, return_pca_vars,
                 vector=False, verbose=None):
    """Get point-spread (PSFs) or cross-talk (CTFs) functions.

    Parameters
    ----------
    resmat : array, shape (n_orient_inv * n_dipoles, n_orient_fwd * n_dipoles)
        Forward Operator.
    src : Source Space
        Source space used to compute resolution matrix.
    %(pctf_idx)s
    func : str ('psf' | 'ctf')
        Whether to produce PSFs or CTFs. Defaults to psf.
    %(pctf_mode)s
    %(pctf_n_comp)s
    %(pctf_norm)s
    %(pctf_return_pca_vars)s
    vector: Bool
        Whether to return PSF/CTF as vector source estimate (3 values per
        location) or source esimate object (1 intensity value per location).
        Only allowed to be True if corresponding dimension of resolution matrix
        is 3 * n_dipoles.
    %(verbose)s

    Returns
    -------
    stcs : instance of SourceEstimate | list of instances of SourceEstimate
    PSFs or CTFs as STC objects.
    All PSFs/CTFs will be returned as successive samples in STC objects,
    in the order they are specified in idx. STCs for different labels will
    be returned as a list.
    If resmat was computed with n_orient_inv==3 for CTFs or n_orient_fwd for
    PSFs then 3 functions per vertex will be returned as successive samples
    (i.e. one function per orientation).
    If vector=False (default) and resmat was computed with n_orient_inv==3 for
    PSFs or n_orient_fwd for CTFs, then the three values per vertex will be
    combined into one intensity value per vertex in a SourceEstimate object.
    If vector=True, PSFs or CTFs with 3 values per vertex (one per orientation)
    will be returned in a VectorSourceEstimate object.
    %(pctf_pca_vars)s
```

#### Additional comments
There could be a separate example to illustrate this option, or I could add something to the existing one (but it would probably require a few figures).

agramfort, 2020-12-07:
no objection

>

larsoner, 2022-02-21:
@olafhauk do you want help here? It would be good if we could take care of not just https://github.com/mne-tools/mne-python/issues/8620 (and the related #10187) but also https://github.com/mne-tools/mne-python/issues/8737

PR #8639 (2022-08-23): ENH: Allow vector source estimate in psf/ctf
#### Reference issue
Closes #8620
Closes #10187

#### What does this implement/fix?
Allows free orientations for get_point_spread()/get_cross_talk(), uses VectorSourceEstimate.

#### Additional information
Next step: Add illustration to existing example for PSFs/CTFs.
closed 2022-08-23 by larsoner