mne-tools/mne-python #8625: plot_sensor_connectivity

BarryLiu-97, 2020-12-08:
When I ran plot_sensor_connectivity.py in cmd, the figure just appeared less than 1 sec, then it was gone.

agramfort, 2020-12-08:
matplotlib figures are non blocking by default so it appears and closes.

you need to use block parameter in matplotlib

closed 2020-12-08 by agramfort

GuillaumeFavelier, 2020-12-08:
Also, can you give more details on how you start the script? Because AFAIK, the 3d plots require an interactive python environment.

So `python -i ./examples/connectivity/plot_sensor_connectivity.py` or `run ./examples/connectivity/plot_sensor_connectivity.py` inside `ipython` works for me.

BarryLiu-97, 2020-12-08:
> Also, can you give more details on how you start the script? Because AFAIK, the 3d plots require an interactive python environment.
> 
> So `python -i ./examples/connectivity/plot_sensor_connectivity.py` or `run ./examples/connectivity/plot_sensor_connectivity.py` inside `ipython` works for me.

just type 'python plot_sensor_connectivity.py'

BarryLiu-97, 2020-12-08:
Ok, it works when I added -i

BarryLiu-97, 2020-12-08:
What does the -i mean?

BarryLiu-97, 2020-12-08:
And is there any way that we can see the channels' name in 3d?

BarryLiu-97, 2020-12-08:
OK, it shows the chan name when I use mayavi

GuillaumeFavelier, 2020-12-08:
From `python --help`:

```
-i     : inspect interactively after running script; forces a prompt even
         if stdin does not appear to be a terminal; also PYTHONINSPECT=x
```

It's normally enabled by default but there seems to be a regression for the `pyvista` 3d backend on the `master` branch. I'll fix it soon :)

BarryLiu-97, 2020-12-08:
Can Miyavi make respond to the mouse click? Like only when I click on the electrode, the name shows up.

GuillaumeFavelier, 2020-12-08:
This feature is not currently implemented.

BarryLiu-97, 2020-12-08:
But could Miyavi support this? Recently I was trying to do a similar task but don't know how to start.

GuillaumeFavelier, 2020-12-08:
You can try opening an issue on [mayavi](https://github.com/enthought/mayavi) to ask directly?

We are transitioning out from `mayavi` in favor of `pyvista` so I doubt such a feature will be added for the `mayavi` 3d backend.

larsoner, 2020-12-08:
I think this should be part of https://github.com/mne-tools/mne-python/issues/8382 -- once we add sensors to the Brain plot, they should be pickable and when you pick them their information + trace can show up in the `show_traces` area.