mne-tools/mne-python #8632: Simulation code (add_ecg, add_eog, ...) overwrites vs sums raw._data structure

jstout211, 2020-12-08:
#### Describe the bug
mne/simulation/raw.py  (add_exg, and add_chpi) overwrite the raw._data matrix.  The final simulated data is the last one performed (because the previous steps have been overwritten).

In mne/simulation/raw.py  (lines 527 and 583 and (possibly 317)) - 
The equals should be replaced with +=
data[picks, start:stop] = einsum('svt,vt->st', fwd, exg_data[:, start:stop])
to
data[picks, start:stop] += einsum('svt,vt->st', fwd, exg_data[:, start:stop])

#### Steps to reproduce
The error can be seen in the example:
https://mne.tools/stable/auto_examples/simulation/plot_simulated_raw_data_using_subject_anatomy.html#sphx-glr-auto-examples-simulation-plot-simulated-raw-data-using-subject-anatomy-py

#### Expected results
[image]

[image]

#### Actual results
[image]

[image]

#### Additional information
Platform:      Linux-4.18.0-240.1.1.el8_3.x86_64-x86_64-with-redhat-8.3-Ootpa
Python:        3.7.9 (default, Aug 31 2020, 12:42:55)  [GCC 7.3.0]
Executable:    /home/stoutjd/miniconda3/envs/MNE_new/bin/python
CPU:           x86_64: 12 cores
Memory:        15.3 GB

mne:           0.21.0
numpy:         1.19.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.3.1 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.1.1
cupy:          Not found
pandas:        1.1.2
dipy:          1.2.0
mayavi:        4.7.3.dev0
pyvista:       0.24.0 {OpenGL 4.6 (Core Profile) Mesa 20.1.4 via Mesa Intel(R) UHD Graphics (CML GT2)}
vtk:           8.2.0
PyQt5:         5.9.2

larsoner, 2020-12-08:
Closing as a duplicate of #8587

Would you be up for making a PR to fix it?

closed 2020-12-08 by larsoner

jstout211, 2020-12-08:
> Closing as a duplicate of #8587
> 
> Would you be up for making a PR to fix it?

Yes - I will try to submit by the end of the day