mne-tools/mne-python #8637: Electrode position coming out of head for topoplot of fNIRS data

Prasenjit1989, 2020-12-09:
#### Describe the bug
Electrode position in topomap for fNIRS data are coming out of head. I have added one image of that. Kindly look into this.
[image]

drammock, 2020-12-09:
Can you replicate this with our [fNIRS motor task](https://mne.tools/dev/overview/datasets_index.html#fnirs-motor) sample data?  If not, can you provide a small dataset that does replicate?  Please also provide a [minimal working example](https://en.wikipedia.org/wiki/Minimal_Working_Example) formatted as [code](https://guides.github.com/features/mastering-markdown/#GitHub-flavored-markdown), that we could copy and paste to replicate the bug.

Prasenjit1989, 2020-12-09:
I replicated the [https://mne.tools/stable/auto_tutorials/preprocessing/plot_70_fnirs_processing.html?highlight=fnirs](url) tutorial for fNIRS motor task data and that is working fine. Actually, our data-set is recorded with earlier version of NIRStar version 14(something like). I contacted the admin of mne as I could not load the data, and I was suggested me to made some correction in the header file to load the data (also done the preprocessing, etc). Now, after epoching, I tried with topoplot, but I am getting such figures. For your convenience, I have attached a sample dataset. Thanks.
[2015-07-28_004.zip](https://github.com/mne-tools/mne-python/files/5667702/2015-07-28_004.zip)

rob-luke, 2020-12-09:
Hi @Prasenjit1989,

Currently MNE only supports NIRx measurements recorded with NIRStar version 15 and above, so 14 is not supported. I suggested a hack to you to get the data to load, but it seems this did not load the locations correctly. Do the positions of optodes also look incorrect in the 3d head imagine when you run the fNIRS demo?

I won't be able to add support for version 14 files. And based on this report it seems that simply forcing the data to load leads to incorrect sensor locations. You could either convert your data to the SNIRF format and load that, or you could add support for V14 to the MNE reader.

Prasenjit1989, 2020-12-11:
Hi Robert,
Thanks for your response. I tried 3D head plot, but getting the following error:

_Traceback (most recent call last):
  File "fnirs_name.py", line 111, in <module>
    fig = mne.viz.plot_alignment(raw.info, show_axes=True, subject='fsaverage', coord_frame='mri',
  File "<decorator-gen-138>", line 21, in plot_alignment
  File "/home/*/anaconda3/envs/mne21/lib/python3.8/site-packages/mne/viz/_3d.py", line 826, in plot_alignment
    surfs[hemi] = read_surface(fname, return_dict=True)[2]
  File "<decorator-gen-71>", line 21, in read_surface
  File "/home/*/anaconda3/envs/mne21/lib/python3.8/site-packages/mne/surface.py", line 718, in read_surface
    fname = _check_fname(fname, 'read', True)
  File "/home/*/anaconda3/envs/mne21/lib/python3.8/site-packages/mne/utils/check.py", line 164, in _check_fname
    raise FileNotFoundError('%s "%s" does not exist' % (name, fname))
FileNotFoundError: File "/data_path/*/2015-07-10_003/fsaverage/surf/lh.pial" does not exist_

Note: 2015-07-10_003 is data folder name.

I think optode positions are stored in _probeInfo.mat file. I checked the source and detector reading section in nirx.py file and  also checked channel location in the epochs (epochs.info[chs]). But, I am clueless where the shifting of optode position is taking place. Anyway, thank you so much.

rob-luke, 2020-12-14:
Hey mate, sorry but NIRx version 14 files arent supported, so its not surprising that the data loads incorrectly. I wont be able to write a new reader for these files, but would be happy to review a PR if you update the reader code. Alternatively you could convert the data to the SNIRF format (probably you would need to go NIRx -> .nirs -> .snirf). If you get this process working please let us know so we can recommend it to others who have old files.

Prasenjit1989, 2021-01-08:
Hi,
Previously we created _probeInfo.mat file using nirsLab. Now, we have created it using NIRSite, and it is working. Anyway, thank you for your suggestions.

larsoner, 2021-01-08:
Glad that there was a suitable workaround!

closed 2021-01-08 by larsoner