mne-tools/mne-python #8640: Tutorial figure is cropped

hoechenberger, 2020-12-10:
I have not checked if I'd get the same results running this locally, but in our rendered docs, I just came across one figure (but there's probably more than this one?) which appears to be cropped:

[image]

(Tutorial: https://mne.tools/dev/auto_tutorials/preprocessing/plot_50_artifact_correction_ssp.html#repairing-ecg-artifacts-with-ssp)

I think this is something we should aim to get fixed before the next release.

compulsory ping @drammock 😅😁

hoechenberger, 2020-12-10:
Also we probably should ensure there's no line break in "PCA" 👌 Not sure if there's an easy way to do this though

mmagnuski, 2020-12-10:
Oh, and the EEG head is huge compared to the MEG heads. :smile:

hoechenberger, 2020-12-10:
> Oh, and the EEG head is huge compared to the MEG heads. 😄

[image]

larsoner, 2020-12-16:
I think we got this in #8651, but if not feel free to reopen

mmagnuski, 2020-12-16:
@larsoner I think this PR fixed the head size inconsistency between MEG and EEG, but not the title being cropped.

larsoner, 2020-12-16:
Ahh okay, I'll reopen. Setting the milestone to 0.23 because I don't think it's a blocker for release

PR #8672 (2020-12-18): VIZ: Fix trellis figure margins
closes #8640 

- Adds a fixed amount (0.25 inch) to the height of `_prepare_trellis` figures to accomodate a title.
- switches `_prepare_trellis` to use the new figure class
- tiny refactor in the `_figure` class functions to be more DRY
- touches a few tutorials to make sure things still look OK
closed 2020-12-18 by larsoner