mne-tools/mne-python #8641: EEGLAB I/O roundtrip fails

hoechenberger, 2020-12-10:
Working on https://github.com/mne-tools/mne-bids/pull/653 I came across a bug when doing an EEGLAB I/O roundtrip.

MWE:
```python
import os.path as op
import mne
from mne.datasets import testing

import mne_bids

data_path = op.join(testing.data_path(), 'EEGLAB')
infile = op.join(data_path, 'test_raw.set')

outfile = '/tmp/foo.set'

raw = mne.io.read_raw_eeglab(infile)
raw.save(outfile, overwrite=True)
mne.io.read_epochs_eeglab(outfile)
```
Produces:
```python
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-1-7ef794074f19> in <module>
     12 raw = mne.io.read_raw_eeglab(infile)
     13 raw.save(outfile, overwrite=True)
---> 14 mne.io.read_epochs_eeglab(outfile)

~/Development/mne-python/mne/io/eeglab/eeglab.py in read_epochs_eeglab(input_fname, events, event_id, eog, verbose, uint16_codec)
    275     .. versionadded:: 0.11.0
    276     """
--> 277     epochs = EpochsEEGLAB(input_fname=input_fname, events=events, eog=eog,
    278                           event_id=event_id, verbose=verbose,
    279                           uint16_codec=uint16_codec)

<decorator-gen-223> in __init__(self, input_fname, events, event_id, tmin, baseline, reject, flat, reject_tmin, reject_tmax, eog, verbose, uint16_codec)

~/Development/mne-python/mne/io/eeglab/eeglab.py in __init__(self, input_fname, events, event_id, tmin, baseline, reject, flat, reject_tmin, reject_tmax, eog, verbose, uint16_codec)
    448                  reject_tmax=None, eog=(), verbose=None,
    449                  uint16_codec=None):  # noqa: D102
--> 450         eeg = _check_load_mat(input_fname, uint16_codec)
    451 
    452         if not ((events is None and event_id is None) or

~/Development/mne-python/mne/io/eeglab/eeglab.py in _check_load_mat(fname, uint16_codec)
     57     """Check if the mat struct contains 'EEG'."""
     58     from ...externals.pymatreader import read_mat
---> 59     eeg = read_mat(fname, uint16_codec=uint16_codec)
     60     if 'ALLEEG' in eeg:
     61         raise NotImplementedError(

~/Development/mne-python/mne/externals/pymatreader/pymatreader.py in read_mat(filename, variable_names, ignore_fields, uint16_codec)
     87                 extra_kwargs['uint16_codec'] = uint16_codec
     88 
---> 89             raw_data = scipy.io.loadmat(fid, struct_as_record=True,
     90                                         squeeze_me=True, mat_dtype=False,
     91                                         variable_names=variable_names,

~/miniforge3/envs/mne/lib/python3.8/site-packages/scipy/io/matlab/mio.py in loadmat(file_name, mdict, appendmat, **kwargs)
    222     with _open_file_context(file_name, appendmat) as f:
    223         MR, _ = mat_reader_factory(f, **kwargs)
--> 224         matfile_dict = MR.get_variables(variable_names)
    225 
    226     if mdict is not None:

~/miniforge3/envs/mne/lib/python3.8/site-packages/scipy/io/matlab/mio4.py in get_variables(self, variable_names)
    389         mdict = {}
    390         while not self.end_of_stream():
--> 391             hdr, next_position = self.read_var_header()
    392             name = asstr(hdr.name)
    393             if variable_names is not None and name not in variable_names:

~/miniforge3/envs/mne/lib/python3.8/site-packages/scipy/io/matlab/mio4.py in read_var_header(self)
    345            position in stream of next variable
    346         '''
--> 347         hdr = self._matrix_reader.read_header()
    348         n = reduce(lambda x, y: x*y, hdr.dims, 1)  # fast product
    349         remaining_bytes = hdr.dtype.itemsize * n

~/miniforge3/envs/mne/lib/python3.8/site-packages/scipy/io/matlab/mio4.py in read_header(self)
    115         O, rest = divmod(rest, 100)  # unused, should be 0
    116         if O != 0:
--> 117             raise ValueError('O in MOPT integer should be 0, wrong format?')
    118         P, rest = divmod(rest, 10)  # data type code e.g miDOUBLE (see above)
    119         T = rest  # matrix type code e.g., mxFULL_CLASS (see above)

ValueError: O in MOPT integer should be 0, wrong format?
```

```
(mne) ➜  mne-python git:(master) mne sys_info
Platform:      macOS-10.16-x86_64-i386-64bit
Python:        3.8.6 | packaged by conda-forge | (default, Nov 27 2020, 19:17:44)  [Clang 11.0.0 ]
Executable:    /Users/hoechenberger/miniforge3/envs/mne/bin/python3.8
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.22.dev0
numpy:         1.19.4 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.5.3
matplotlib:    3.3.3 {backend=MacOSX}

sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.2.1
cupy:          Not found
pandas:        1.1.4
dipy:          1.3.0
mayavi:        4.7.2
pyvista:       0.27.3 {pyvistaqt=0.2.0, OpenGL 4.1 INTEL-16.0.49 via Intel(R) Iris(TM) Plus Graphics 655}
vtk:           9.0.1
PyQt5:         5.12.3
```

hoechenberger, 2020-12-10:
Obviously… you shouldn't try to read raw data as epochs, right? Ugh… sorry.

hoechenberger, 2020-12-10:
No, wait. Even with the corrected MWE, I get the same error.

```python
import os.path as op
import mne
from mne.datasets import testing

data_path = op.join(testing.data_path(), 'EEGLAB')
infile = op.join(data_path, 'test_raw.set')

outfile = '/tmp/foo.set'

raw = mne.io.read_raw_eeglab(infile)
raw.save(outfile, overwrite=True)
mne.io.read_raw_eeglab(outfile)
```

cc @cbrnr

hoechenberger, 2020-12-10:
ok. we're producing a FIFF file. No wonder we cannot read it :)

closed 2020-12-10 by hoechenberger

cbrnr, 2020-12-10:
While you're at it, could you test if EEGLAB export in `mnelab.io.write_raw` is working as expected? Then at least one example is actually doing what you wanted 😛.

hoechenberger, 2020-12-10:
I'm struggling ;)