mne-tools/mne-python #8642: BUG: stc plot color limits

kingjr, 2020-12-10:
@larsoner 000dac95c0f7a4439ee60fb92f1007fd288f9611 introduced a color limit bug in the stc plot

```python
import numpy as np
from pathlib import Path
from mayavi import mlab
import matplotlib.pyplot as plt

import mne

data_path = Path(mne.datasets.sample.data_path())
fname = data_path / 'MEG' / 'sample' / 'sample_audvis-meg'

# Read stc from file
stc = mne.read_source_estimate(fname, subject='sample')
stc.data[:, :] = np.random.rand(*stc.data.shape) # some positive data

fig_size = (300, 300)
fig = mlab.figure(size=fig_size)

brain = stc.plot(
    hemi='lh',
    views='lat',
    colormap='coolwarm',
    subjects_dir=data_path / 'subjects',
    clim=dict(kind='value', pos_lims=[0, .2, .5]),  # <-------------------- this what breaks
    backend='mayavi',
    figure=fig,
)

img = mlab.screenshot(figure=fig, mode='rgba', antialiased=True)
plt.imshow(img)
```

before:
[image]

after:
[image]

kingjr, 2020-12-10:
cc @GuillaumeFavelier who probably know intimately the intricate 3d plotting functions

larsoner, 2020-12-10:
I can replicate on latest `master`, @GuillaumeFavelier I can make a quick PR for this since I just dumped https://github.com/mne-tools/mne-python/pull/8335#issuecomment-742531087 on you :)

closed 2020-12-14 by GuillaumeFavelier

NeuroLaunch, 2020-12-14:
Nice catch, as I was just encountering this bug.

PR #8644 (2020-12-14): MRG, BUG: Mayavi center
Closes #8642

Feel free to try it @kingjr . But you should really switch to using the PyVista backend and complain about problems there and we will fix them :)