mne-tools/mne-python #8646: Suggestion: find_labels_adjacency() function for cortical labels

darthvaderismymom, 2020-12-11:
#### Describe the new feature or enhancement
Apparently, among functions that compute adjacency matrix for channels and sources, MNE currently does not have a function that computes the spatial adjacency matrix for cortical labels (e.g., Desikan-Killiani atlas with 68 cortical labels). Similar to mne.channels.find_ch_adjacency() function, such function should have list of labels as input and return the scipy.sparse.csr_matrix of labels adjacency with shape (n_labels, n_labels). The output of such function could be used for cluster permutational statistics.

#### Describe your proposed implementation
I suggest it can be done relatively easy using FreeSurfer triangulations.

lqxxgnd, 2024-09-10:
i agree with you!

larsoner, 2024-09-13:
Agreed this would be good, if someone is up for giving it a shot feel free to open a PR!

wmvanvliet, 2026-08-25:
Here is a gist which an implementation: https://gist.github.com/wmvanvliet/fc0f1a3334955698303841fcc8a8d85d

wmvanvliet, 2026-08-25:
@raphbrd, @payamsash

wmvanvliet, 2026-08-25:
If we want to support volumetric labels,we may first have to create a `VolLabel` class, because right now all the functions dealing with volumetric labels take an `aseg`/`labels` pair of arguments.

larsoner, 2026-08-25:
Yeah for the first PR I would just support handling a cortical parcellation and/or list of cortical surface labels. Volume could come later

closed 2026-08-27 by larsoner