mne-tools/mne-python #8650: IndexError when calling more then one epochs.plot_psd()

LiamMBooth, 2020-12-11:
#### Describe the bug
When calling a second plot_psd() in the same script on my epochs object i get an error from scipy.signal.windows.windows.py,

I suspect it may be something to do with my current dependencies, so you're aware:
Windows 10,
Python 3.7.3
MNE: 0.21.2
numpy: 1.19.3
scipy: 1.5.0

#### Steps to reproduce
```
import mne
import matplotlib.pyplot as plt
from mne.preprocessing import (ICA, create_eog_epochs, create_ecg_epochs,corrmap)

raw = mne.io.read_raw_fif('raw.fif')
events = mne.find_events(raw, stim_channel='Status')

mapping = {'Visual/easy':1,'Visual/hard':2,'Audiotry/easy':3,
           'Auditory/hard':4,'V/E':5,'V/H':6,'A/E':7,'A/H':8,
           'Start':9,'Stop':10}

epochs = mne.Epochs(raw, events, tmin=-0.2, tmax=0.5, event_id=mapping, preload=True)

epochs['Visual/easy'].plot_psd() 
epochs['Visual/hard'].plot_psd()
```

#### Expected results
I expect a second plot to open with the psd for my epochs of Visual/hard, i successfully get my psd plot for Visual/easy on my first call, this error is only thrown on my second call.

#### Actual results
```
C:\Users\Liam\Documents\python\EEG_process>math2.py
Opening raw data file raw.fif...
    Range : 0 ... 699629 =      0.000 ...  2798.516 secs
Ready.
486 events found
Event IDs: [ 1  2  3  4  5  6  7  8  9 10]
Not setting metadata
Not setting metadata
486 matching events found
Applying baseline correction (mode: mean)
0 projection items activated
Loading data for 486 events and 176 original time points ...
0 bad epochs dropped
    Using multitaper spectrum estimation with 7 DPSS windows
Traceback (most recent call last):
  File "C:\Users\Liam\Documents\python\EEG_process\math2.py", line 15, in <module>
    epochs['Visual/hard'].plot_psd()
  File "C:\python37\lib\site-packages\mne\epochs.py", line 1096, in plot_psd
    verbose=verbose)
  File "<decorator-gen-147>", line 24, in plot_epochs_psd
  File "C:\python37\lib\site-packages\mne\viz\epochs.py", line 955, in plot_epochs_psd
    n_jobs=n_jobs)
  File "<decorator-gen-65>", line 24, in psd_multitaper
  File "C:\python37\lib\site-packages\mne\time_frequency\psd.py", line 330, in psd_multitaper
    n_jobs=n_jobs, verbose=verbose)
  File "<decorator-gen-50>", line 24, in psd_array_multitaper
  File "C:\python37\lib\site-packages\mne\time_frequency\multitaper.py", line 422, in psd_array_multitaper
    n_times, sfreq, bandwidth, low_bias, adaptive)
  File "<decorator-gen-49>", line 24, in _compute_mt_params
  File "C:\python37\lib\site-packages\mne\time_frequency\multitaper.py", line 350, in _compute_mt_params
    interp_from=interp_from)
  File "C:\python37\lib\site-packages\mne\time_frequency\multitaper.py", line 95, in dpss_windows
    dpss = _get_dpss()(N, half_nbw, Kmax)
  File "C:\python37\lib\site-packages\scipy\signal\windows\windows.py", line 1938, in dpss
    if w[w * w > thresh][0] < 0:
IndexError: index 0 is out of bounds for axis 0 with size 0
```

C:\Users\Liam\Documents\python\EEG_process>

#### Additional information
```
Platform:      Windows-10-10.0.19041-SP0
Python:        3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 22:22:05) [MSC v.1916 64 bit (AMD64)]
Executable:    C:\python37\python.exe
CPU:           Intel64 Family 6 Model 60 Stepping 3, GenuineIntel: 4 cores
Memory:        31.9 GB

mne:           0.21.2
numpy:         1.19.3 {blas=D:\\a\\1\\s\\numpy\\build\\openblas_info, lapack=D:\\a\\1\\s\\numpy\\build\\openblas_lapack_info}
scipy:         1.5.0
matplotlib:    3.2.1 {backend=TkAgg}

sklearn:       0.21.2
numba:         Not found
nibabel:       Not found
cupy:          Not found
pandas:        0.25.3
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found
```

agramfort, 2020-12-12:
can you replicate with the sample data?

are you sure epochs['Visual/hard'] is not empty?

>

LiamMBooth, 2020-12-12:
Hi Alexandre, thank you for your response.

If i call epochs['Visual/hard'].plot_psd() before  epochs['Visual/easy'].plot_psd() it yields a plot, but again has the same failure on the second plot_psd() function, the same result happens if i repeat the function on the same epoch with success on the first plot and indexerror on the second.

I've recreated with sample datasets using code from the example given in [this link](https://mne.tools/dev/auto_tutorials/intro/plot_10_overview.html#sphx-glr-auto-tutorials-intro-plot-10-overview-py):
```
import mne
import os
import matplotlib.pyplot as plt
from mne.preprocessing import (ICA, create_eog_epochs, create_ecg_epochs,corrmap)

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_filt-0-40_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)

event_dict = {'auditory/left': 1, 'auditory/right': 2, 'visual/left': 3,
              'visual/right': 4, 'smiley': 5, 'buttonpress': 32}

events = mne.find_events(raw, stim_channel='STI 014')

epochs = mne.Epochs(raw, events, event_id=event_dict, tmin=-0.2, tmax=0.5, preload=True)
                    
epochs['auditory/left'].plot_psd()
epochs['auditory/left'].plot_psd()         
```

And my prompt is similar to before:
```
C:\Users\Liam\Documents\python\EEG_process>math2.py
Opening raw data file C:\Users\Liam\mne_data\MNE-sample-data\MEG\sample\sample_audvis_filt-0-40_raw.fif...
    Read a total of 4 projection items:
        PCA-v1 (1 x 102)  idle
        PCA-v2 (1 x 102)  idle
        PCA-v3 (1 x 102)  idle
        Average EEG reference (1 x 60)  idle
    Range : 6450 ... 48149 =     42.956 ...   320.665 secs
Ready.
319 events found
Event IDs: [ 1  2  3  4  5 32]
Not setting metadata
Not setting metadata
319 matching events found
Applying baseline correction (mode: mean)
Created an SSP operator (subspace dimension = 4)
4 projection items activated
Loading data for 319 events and 106 original time points ...
0 bad epochs dropped
    Using multitaper spectrum estimation with 7 DPSS windows
    Using multitaper spectrum estimation with 7 DPSS windows
    Using multitaper spectrum estimation with 7 DPSS windows
Traceback (most recent call last):
  File "C:\Users\Liam\Documents\python\EEG_process\math2.py", line 19, in <module>
    epochs['auditory/left'].plot_psd()
  File "C:\python37\lib\site-packages\mne\epochs.py", line 1096, in plot_psd
    verbose=verbose)
  File "<decorator-gen-147>", line 24, in plot_epochs_psd
  File "C:\python37\lib\site-packages\mne\viz\epochs.py", line 955, in plot_epochs_psd
    n_jobs=n_jobs)
  File "<decorator-gen-65>", line 24, in psd_multitaper
  File "C:\python37\lib\site-packages\mne\time_frequency\psd.py", line 330, in psd_multitaper
    n_jobs=n_jobs, verbose=verbose)
  File "<decorator-gen-50>", line 24, in psd_array_multitaper
  File "C:\python37\lib\site-packages\mne\time_frequency\multitaper.py", line 422, in psd_array_multitaper
    n_times, sfreq, bandwidth, low_bias, adaptive)
  File "<decorator-gen-49>", line 24, in _compute_mt_params
  File "C:\python37\lib\site-packages\mne\time_frequency\multitaper.py", line 350, in _compute_mt_params
    interp_from=interp_from)
  File "C:\python37\lib\site-packages\mne\time_frequency\multitaper.py", line 95, in dpss_windows
    dpss = _get_dpss()(N, half_nbw, Kmax)
  File "C:\python37\lib\site-packages\scipy\signal\windows\windows.py", line 1938, in dpss
    if w[w * w > thresh][0] < 0:
IndexError: index 0 is out of bounds for axis 0 with size 0
```

agramfort, 2020-12-12:
this code works great for me on master and 0.21 branch.

can someone else try? could be a windows issue

>

LiamMBooth, 2020-12-13:
UPDATE: No luck, I've just tried uninstalling mne, scipy and numpy and reinstalling them all in case i've edited something i shouldn't have over the past few months i may have forgotten about and I am still getting the same response.

agramfort, 2020-12-14:
could someone with a windows machine replicate?

larsoner, 2020-12-17:
On my Windows 10 VM I did:
```
$ curl -O https://raw.githubusercontent.com/mne-tools/mne-python/master/environment.yml
$ conda install --name base nb_conda_kernels
$ conda env update --file environment.yml
$ conda activate mne
$ mne sys_info
Platform:      Windows-10-10.0.19041-SP0
Python:        3.8.6 | packaged by conda-forge | (default, Nov 27 2020, 18:58:29) [MSC v.1916 64 bit (AMD64)]
Executable:    C:\Users\tester\Miniconda3\envs\mne\python.exe
CPU:           Intel64 Family 6 Model 158 Stepping 9, GenuineIntel: 4 cores
Memory:        5.7 GB

mne:           0.21.2
numpy:         1.19.4 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.5.3
matplotlib:    3.3.3 {backend=Qt5Agg}

sklearn:       0.23.2
numba:         0.52.0
nibabel:       3.2.1
cupy:          Not found
pandas:        1.1.5
dipy:          1.3.0
mayavi:        4.7.2
pyvista:       0.27.4 {pyvistaqt=0.2.0, OpenGL 3.3 (Core Profile) Mesa 10.2.4 (git-d92815a) via Gallium 0.4 on llvmpipe (LLVM 3.4, 256 bits)}
vtk:           9.0.1
PyQt5:         5.12.3
```
I cannot replicate. It's probably a numpy/scipy BLAS issue, possibly related to the CPU hardware and/or Windows (they have broken some stuff lately).

Maybe try `pip install --upgrade --pre numpy`? It should pull 1.20rc1, which might have some fixes.

LiamMBooth, 2020-12-17:
I've gone to Python 3.8.6 and I tried installing numpy to 1.19.4 and got an error which would hint strongly to your comment @larsoner:
`RuntimeError: The current Numpy installation ('C:\\python38\\lib\\site-packages\\numpy\\__init__.py') fails to pass a sanity check due to a bug in the windows runtime. See this issue for more information: https://tinyurl.com/y3dm3h86`

I've just tried `pip install --upgrade --pre numpy`, got the same result, then i tried  `pip install --upgrade --pre scipy` still no luck

Is this an issue more suited to numpy/scipy on windows machines? For now i'll pull out the raw epoch data to hack at in numpy/scipy.

larsoner, 2020-12-17:
Can you try `pip install --upgrade --pre numpy`? It should use NumPy `1.20rc1`

LiamMBooth, 2020-12-17:
I have done, that's the current version i have installed:
`C:\Users\Liam\Documents\python\EEG_process>python
Python 3.8.6 (tags/v3.8.6:db45529, Sep 23 2020, 15:52:53) [MSC v.1927 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import mne
>>> mne.sys_info()
Platform:      Windows-10-10.0.19041-SP0
Python:        3.8.6 (tags/v3.8.6:db45529, Sep 23 2020, 15:52:53) [MSC v.1927 64 bit (AMD64)]
Executable:    C:\python38\python.exe
CPU:           Intel64 Family 6 Model 60 Stepping 3, GenuineIntel: 4 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.21.2
numpy:         1.20.0rc1 {blas=D:\\a\\1\\s\\numpy\\build\\openblas_info, lapack=D:\\a\\1\\s\\numpy\\build\\openblas_lapack_info}
scipy:         1.6.0rc1
matplotlib:    3.3.3 {backend=TkAgg}

sklearn:       Not found
numba:         Not found
nibabel:       Not found
cupy:          Not found
pandas:        Not found
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found`

drammock, 2020-12-17:
after reading through a good chunk of the issue linked in the NumPy RuntimeError message, it sounds like:

1. it's a windows problem. changing numpy versions won't fix it; the difference between 1.19.3 and 1.19.4 is that 1.19.4 warns you about it in advance. Here is [a direct link](https://developercommunity.visualstudio.com/comments/1243113/view.html) to the comment by the MSFT dev explaining the underlying issue.
2. the timeline for a windows fix to be released is sometime in **late January**
3. in the meantime, the suggested workaround is:
    - set up and use a 32-bit python environment for now. Only 64-bit libraries are affected
    - in that 32-bit environment, pin to `numpy==1.19.3` to avoid getting the "Sanity check" warning

LiamMBooth, 2020-12-17:
Ahhh it all makes sense now, i was convinced it was something to do with SciPy, moving to 32-bit environment will work fine for me for now, I'll come back to try again in a 64-bit environment on Windows in a couple of months to see if its fixed. 

I should probably start exploring with linux or some other development environments but i'm so accustomed to doing everything in command prompt and notepad++.

Thankyou all for your comments and guidance, i'll report when i've installed and tried again.

LiamMBooth, 2020-12-17:
Perfect, seems to be working great in a 32-bit python environment, i'll close the issue now. 

Thanks again!

closed 2020-12-17 by LiamMBooth