mne-tools/mne-python #8660: PSDs computed with psd_welch look different between 0.20 and 0.21

nschawor, 2020-12-15:
#### Describe the bug

I noticed that the output of ``` mne.time_frequency.psd_welch``` looks slightly different between versions, example for a sine wave:

[image]

#### Steps to reproduce
Here is a minimum working example with simulated data: 

```python
import mne
import numpy as np
import matplotlib.pyplot as plt

# some simulated signal
frequency = 10
dt = 0.001
fs = 1 / dt
n_seconds = 10
time = np.arange(0, n_seconds, dt)
signal = np.cos(2 * np.pi * frequency * time)

info = mne.create_info(["signal1"], fs, ch_types="eeg")
raw = mne.io.RawArray(signal[np.newaxis], info)

# compute PSD
psd, freq = mne.time_frequency.psd_welch(raw, fmin=1, fmax=30, n_fft=4000)

# plot
fig, ax = plt.subplots()
ax.semilogy(freq, psd.T)
fig.show()
```

#### Expected results
I expected it to look the same between versions. Is this a desired change? I liked the old behavior. :-) Can it be achieved with the new version somehow? I checked the change log for 0.21 where optimization of ```psd_welch``` for long data segments is reported, which may have led to this?

#### Additional information
I tested various MNE versions and the change appears between 0.20.8 and 0.21.0 and persists to 0.21.2.

cbrnr, 2020-12-15:
Indeed #7994 is the culprit - @larsoner or @hoechenberger do you want to take a look?

On a related note (and I'm sure I've probably asked this before), which window do we use and how can we change it?

hoechenberger, 2020-12-15:
@cbrnr I'll leave this to @larsoner 😁😅

PR #8662 (2020-12-15): MRG, BUG, ENH: Add window option
Closes #8660
closed 2020-12-15 by cbrnr