mne-tools/mne-python #8664: The MNE website is really slow to access from China

sakimarquis, 2020-12-16:
I don't whether this is a suitable problem to issue here. However, I can't find other ways to report this problem. The problem has last for a long time, and recently the speed is intolerably slow. I ask several friends to test from different regions in China, the problem persists. And when I access the MNE website by VPN, the speed improves.

agramfort, 2020-12-16:
the website is hosted on github. You could mirror the website in china if
you want

>

drammock, 2020-12-16:
The problem is likely to be (at least partially) calls made to the Google API. We could consider ditching Google analytics, and see if we make any other Google API calls (we don't use Google fonts, at least)

hoechenberger, 2020-12-17:
> The problem is likely to be (at least partially) calls made to the Google API. We could consider ditching Google analytics, and see if we make any other Google API calls (we don't use Google fonts, at least)

The API calls are asynchronous, i.e. non-blocking. I suspect it's the JS source file that is loaded from Google's CDN. We could simply integrate it into our docs (i.e., serve it from GH pages). This should fix the issue.

larsoner, 2020-12-17:
@hoechenberger can you open a PR to integrate the JS file? I'm not sure how to keep it up to date. Maybe just a file in `_static` that we update as part of the release process?

hoechenberger, 2020-12-17:
> @hoechenberger can you open a PR to integrate the JS file? I'm not sure how to keep it up to date. Maybe just a file in `_static` that we update as part of the release process?

Yes, that'd be worth a shot. Or we actually set up a CI task that runs every night or once a week or so that retrieves the file and commits it to `master`.

**IF** that fixes @sakimarquis's issue.

agramfort, 2020-12-17:
let's check first it's the cause of the problem

>

hoechenberger, 2020-12-17:
@sakimarquis Does https://24173-1301584-gh.circle-artifacts.com/0/dev/index.html load any faster for you?

sakimarquis, 2020-12-18:
> @sakimarquis Does https://24173-1301584-gh.circle-artifacts.com/0/dev/index.html load any faster for you?

Unfortunately no. Overall they cost relatively same time for load the whole page. This result is tested in different browser and ISPs.

larsoner, 2023-10-01:
Closing in favor of https://github.com/mne-tools/mne-python/issues/11881 which I think is all we can do to try to fix this

closed 2023-10-01 by larsoner