mne-tools/mne-python #8666: Source covariance Matrix?

LF9076, 2020-12-17:
how can i change the source Covariance matrix in the inverse problem ?theThe minimum-norm current estimates said that The amplitudes of the currents have a Gaussian prior distribution with a known source covariance matrix.I am not very clear about this? The source covariance matrix can be I(Identity matrix) if there are no prior information? Thanks a lot ~~

agramfort, 2020-12-17:
did you read https://mne.tools/stable/overview/implementation.html?highlight=dspm#the-minimum-norm-current-estimates ?

larsoner, 2020-12-17:
I don't think we currently expose a parameter in `make_inverse_operator` to allow actually setting the *a priori* source covariance matrix, i.e., something similar to the MNE-C options:
```
	--srccov name      Specify the source covariance matrix (defaults to identity matrix)
	--fmri name        Specify the fMRI weighting file (w format)
	--fmrithresh val   Specify the threshold for the fMRI weighting
	--fmrioff    val   Specify the source variance value in locations with no fMRI activation ( 0.10)
```
In theory we could add a `source_cov : None | ndarray, shape (n_vertices * n_orientations,)` parameter. The default (None) is equvialent to `np.ones(n_vertices * n_orientations)`, then whatever you give gets weighted according to `depth` and `loose` as usual. In principle this shouldn't be too difficult to add.

Is this what you're looking for @LF9076 ?

larsoner, 2020-12-17:
(in principle we could add support for full or block-diagonal-3x3 source covariance matrices, too, but at least starting with the diagonal case would be simple enough)

LF9076, 2020-12-17:
> did you read https://mne.tools/stable/overview/implementation.html?highlight=dspm#the-minimum-norm-current-estimates ?

Thanks a lot ,i have read about it clearly~~

LF9076, 2020-12-17:
> 我认为我们目前没有公开`make_inverse_operator`允许实际设置_先验_源协方差矩阵的参数，即类似于MNE-C选项的东西：
> 
> ```
> 	--srccov name      Specify the source covariance matrix (defaults to identity matrix)
> 	--fmri name        Specify the fMRI weighting file (w format)
> 	--fmrithresh val   Specify the threshold for the fMRI weighting
> 	--fmrioff    val   Specify the source variance value in locations with no fMRI activation ( 0.10)
> ```
> 
> 从理论上讲，我们可以添加一个`source_cov : None | ndarray, shape (n_vertices * n_orientations,)`参数。默认值（None）等价于`np.ones(n_vertices * n_orientations)`，然后您提供的任何内容都会按照`depth`和`loose`照常进行加权。原则上，添加起来应该不太困难。
> 
> 这是您在寻找@ LF9076的东西吗？
So thanks, in MNE-C there can change the souce_cov due to fMRI information? actually i got some fNIR signals ,so fNIR is same to fMRI signals?

ttaiv, 2026-09-08:
@larsoner I would be interested in adding the `source_cov` parameter. We would like to use it to address the different densities of cortical and cerebellar meshes (output of the [CMB](https://github.com/johnsam7/ceremegbellum) package, related PR [here](https://github.com/mne-tools/mne-python/pull/14130)). The idea is to set prior source variances to be proportional to the area represented by each source. What do you think?

larsoner, 2026-09-08:
That seems reasonable. Might be better if we could somehow make the spacing obtained from the subcortical sources match that of the cortical. But if we can't (and even if we only can approximately) it seems reasonable to have a function that computed a source cov that would help normalize.

ttaiv, 2026-09-09:
To make the spacing of cortical and subcortical source spaces match, `setup_subcortical_source_space` should probably take a `spacing` parameter. But I guess arbitrary `source_cov` could be useful for some users and not too hard to implement.