mne-tools/mne-python #8671: Reading Data from QuSpin Gen 2 OPMs

neurofractal, 2020-12-17:
Hi all. Planning on writing some code for importing data acquired using QuSpin OPMs: https://quspin.com/products-qzfm/ . A question before I start:

There is currently no data saving standard for this device. Our lab has opted to save the data in a simple .bin format and add BIDS descriptor json/tsv files. OK to implement for this specific use-case? I don't see too many issues given the limited number of teams using OPMs right now...

agramfort, 2020-12-18:
this is no acq system to allow to save your data for such device?

>

larsoner, 2020-12-18:
@LauriParkkonen and @rzetter added a QuSpin OPM example back in https://github.com/mne-tools/mne-python/pull/5525 where the data was in FIF format. Any thoughts on QuSpin formats / acquisition / saving? Was writing to FIF a home-grown solution?

LauriParkkonen, 2020-12-18:
The plain QuSpin software cannot save the data properly - it gets 
excerpts of the data via USB and is for monitoring/viewing only. Thus, 
one has to hook up a proper data acquisition system to the QuSpin sensor 
electronics. We used a customized version of the MEGIN/Elekta/Neuromag 
SQUID MEG system DACQ and the associated software, which directly 
produces FIFF files.

Eric Larson kirjoitti 18.12.2020 klo 14.14:

larsoner, 2020-12-18:
Maybe a good use case for [MNE-SCAN](https://pubmed.ncbi.nlm.nih.gov/29621570/)?

@neurofractal is changing your acquisition method to something more standard / flexible another option, or are the plain binaries preferred at your site(s)? (And good to see you on GitHub again :) )

neurofractal, 2020-12-18:
Hey @larsoner - good to be contributing to MNE again! Our lab is going to stick with binaries for the moment. I can implement and then other groups can customise with their solutions?

larsoner, 2020-12-18:
Do you get the binaries in some semi-standard fashion from the system? I'm worried about including something too lab-specific in MNE -- for example if you have an in-house (and not used / planned to be used anywhere else) data-saving scheme then it might make sense to keep an in-house data-reading scheme, as well, at least to start. @bloyl do you have some insights from your work with the Artemis123 system?

How close is your data from being usable directly with RawArray + create_info? I'm guessing the difficulty is mostly with the sensor positions, but maybe we can come up with a solution for that. For example if this is all you need:
```
info = mne.create_info(25, 1000., 'mag')
for ch in info ['chs]':
    ch['coil_type'] = ...
data = np.fromfile(..., dtype=np.float64).reshape(...)
raw = mne.io.RawArray(data, info)
```
then the "in-house" wrapper isn't too bad. This doesn't implement things like on-demand reads that you would get from having a proper BaseRaw instance, but those could be added fairly easily I think.

neurofractal, 2020-12-18:
@larsoner OK - sounds like a plan. I have already made something very similar, but the main issues were with sensor positions, orientations and some weird scaling.

Where would such a wrapper sit within MNE?

rzetter, 2020-12-18:
@neurofractal Just in case if you're using the USB serial stream for your binary files, be advised that it's only at 200 Hz. Further, the serial stream data are not time-stamped, meaning that any latencies in the data transfer (e.g. when using multiple sensors and a USB hub, or if the measurement PC is under heavy load) may affect the data. Finally, IIRC the data has low resolution (bit depth). If you're already using the analog output (e.g. with the NI dacq system that Quspin sells https://quspin.com/products-qzfm/ni-9205-data-acquisition-unit/), the approach discussed sounds great!

Sensor positions is an issue, and to my understanding (I might be wrong) BIDS isn't quite suitable for (wearable) MEG data in which the sensor positions (relative to each other) are variable across subjects using the same MEG system. Something like the EEG-BIDS *_electrodes.tsv field would perhaps be required?

neurofractal, 2020-12-18:
Thanks @rzetter 

1. Yep using the NI dacq
2. Exactly. We make a ..._channels.tsv file for each recording with position(xyz) and orientation (xyz) as extra columns
3. Perfect - that will be very useful 👍

larsoner, 2023-05-24:
I think we have this now but feel free to reopen if I'm mistaken (thanks @georgeoneill !)

dbobela406, 2024-09-27:
Hi All, 

Just echoing what Robert mentioned. I just posted some QuSpin Neuro-1 sample data with a document that describes the formatting [here](https://github.com/fieldtrip/website/issues/807). This may help with the use case of data saved using the QuSpin Neuro-1 software (labview based). We're finishing up some TCP/IP data stream details, but once those are finalized, I'll provide a document explaining the data stream format. Reach out any time with questions, cheers!

larsoner, 2024-09-30:
Okay I'll reopen this since from a quick look this looks like a new file format. So we either need to improve our existing reader or write a new one. Not sure which is the best option. Would depend on how similar the formats are probably and someone will need to dig into the details a bit to know.

georgeoneill, 2024-09-30:
I'm not directly based in the UCL OPM lab anymore (now a 3rd party user). But we have a Neuro-1 here so can get some more information on the state of the file format. My hunch is QuSpin use the .lvm from LabView and then we decorate it with our BIDS-like metadata offline. But let me get back to you on that.

neurofractal, 2024-10-01:
Happy to be involved in writing the reader for this @larsoner @dbobela406 - I'll be using Neuro-1 in combination with MNE very soon! I'll follow the discussion on the Fieldtrip post as well. OK to open a new issue as this isn't for Gen-2 OPMs...?

closed 2024-10-01 by larsoner