mne-tools/mne-python #8677: typerror in make_dics when using the csd as input

EPinzuti, 2020-12-21:
I pass to make_dics the output of csd_multitaper  and I get this typerror 

 File "/data/home1/epinzuti/anaconda3/lib/python3.7/site-packages/mne/beamformer/_dics.py", line 174, in make_dics
    info, fwd, csd = equalize_channels([info, forward, csd])

  File "<decorator-gen-31>", line 24, in equalize_channels

  File "/data/home1/epinzuti/anaconda3/lib/python3.7/site-packages/mne/channels/channels.py", line 148, in equalize_channels
    allowed_types_str)

  File "/data/home1/epinzuti/anaconda3/lib/python3.7/site-packages/mne/utils/check.py", line 389, in _validate_type
    % (item_name, type_name, type(item),))

TypeError: Instances to be modified must be an instance of Raw, Epochs, Evoked, TFR, Forward, Covariance, CrossSpectralDensity or Info, got <class 'mne.time_frequency.csd.CrossSpectralDensity'> instead

agramfort, 2020-12-21:
@EPinzuti please make an effort when you open an issue to follow the template we took the time to write:

https://github.com/mne-tools/mne-python/issues/new?assignees=&labels=BUG&template=bug_report.md&title=

you don't provide any code snippet, etc.

thanks for your understanding

drammock, 2020-12-21:
duplicate of #8678

closed 2020-12-21 by drammock