mne-tools/mne-python #8680: Error of Incorrect Sample number when loading .set with preloading=True

Aceticia, 2020-12-24:
#### Describe the bug
When using `mne.io.read_raw_eeglab` on a public dataset's '.set' files, I get this error:
`RuntimeError: Incorrect number of samples (5505024 != 23104512), please report this error to MNE-Python developers`

#### Steps to reproduce
I downloaded the data from this dataset: https://openneuro.org/datasets/ds002893/versions/1.0.0, and the '.set' files give this error.

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
Expected behavior is that files are loaded without error.

#### Actual results
```Traceback (most recent call last):
  File "dataset.py", line 669, in <module>
    data = AffectiveMusicListening(64, '../data/', verbose=True)
  File "dataset.py", line 630, in __init__
    super().__init__(electrode_lst, xyz_steps, directory, verbose=verbose)
  File "dataset.py", line 360, in __init__
    electrode_loc_temp.append(locs.loc[elec_name].to_numpy())
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/pandas/core/indexing.py", line 879, in __getitem__
    return self._getitem_axis(maybe_callable, axis=axis)
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/pandas/core/indexing.py", line 1110, in _getitem_axis
    return self._get_label(key, axis=axis)
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/pandas/core/indexing.py", line 1059, in _get_label
    return self.obj.xs(label, axis=axis)
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/pandas/core/generic.py", line 3493, in xs
    loc = self.index.get_loc(key)
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/pandas/core/indexes/base.py", line 2900, in get_loc
    raise KeyError(key) from err
```

#### Additional information
```Platform:      Linux-3.10.0-1062.18.1.el7.x86_64-x86_64-with-redhat-7.7-Maipo
Python:        3.7.9 (default, Aug 31 2020, 12:42:55)  [GCC 7.3.0]
Executable:    /afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/bin/python
CPU:           x86_64: 20 cores
Memory:        19.4 GB

mne:           0.21.2
numpy:         1.19.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.2
matplotlib:    3.3.2 {backend=agg}

sklearn:       Not found
numba:         Not found
nibabel:       Not found
cupy:          Not found
pandas:        1.1.5
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found```

agramfort, 2020-12-25:
I cannot replicate:

In [6]: import mne

In [7]: raw =
mne.io.read_raw('sub-001_eeg_sub-001_task-AuditoryVisualShift_run-01_eeg.set')
<ipython-input-7-2582adc85011>:1: RuntimeWarning: Data file name in
EEG.data (sub-001_task-AuditoryVisualShift_run-01_eeg.fdt) is incorrect,
the file name must have changed on disk, using the correct file name
(sub-001_eeg_sub-001_task-AuditoryVisualShift_run-01_eeg.fdt).
  raw =
mne.io.read_raw('sub-001_eeg_sub-001_task-AuditoryVisualShift_run-01_eeg.set')

please share a full code snippet to replicate the issue.
Maybe it's fixed in version 0.22

>

Aceticia, 2020-12-30:
Sorry for late replying, can use this code to reproduce it:
```
mne.io.read_raw_eeglab(
        'OpenNeuro/AudioVisualShift/sub-004/eeg/sub-004_task-AuditoryVisualShift_run-02_eeg.set',
        preload=True)
```
You might get this error:
```
Reading ../data/OpenNeuro/AudioVisualShift/sub-004/eeg/sub-004_task-AuditoryVisualShift_run-02_eeg.fdt
Reading 0 ... 641791  =      0.000 ...  2566.863 secs...
Traceback (most recent call last):
  File "dataset.py", line 885, in <module>
    preload=True)
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/mne/io/eeglab/eeglab.py", line 222, in read_raw_eeglab
    eog=eog, verbose=verbose, uint16_codec=uint16_codec)
  File "<decorator-gen-216>", line 24, in __init__
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/mne/io/eeglab/eeglab.py", line 335, in __init__
    orig_format='double', verbose=verbose)
  File "<decorator-gen-161>", line 24, in __init__
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/mne/io/base.py", line 249, in __init__
    self._preload_data(preload)
  File "<decorator-gen-165>", line 24, in _preload_data
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/mne/io/base.py", line 531, in _preload_data
    data_buffer=data_buffer, projector=self._projector)
  File "<decorator-gen-163>", line 24, in _read_segment
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/mne/io/base.py", line 416, in _read_segment
    int(start_file), int(stop_file), cals, mult)
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/mne/io/base.py", line 1813, in _read_segment_file
    self, data, idx, fi, start, stop, cals, mult)
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/mne/io/eeglab/eeglab.py", line 367, in _read_segment_file
    self, data, idx, fi, start, stop, cals, mult, dtype='<f4')
  File "/afs/ece.cmu.edu/usr/xujinl/anaconda3/envs/FNO/lib/python3.7/site-packages/mne/io/utils.py", line 224, in _read_segments_file
    'developers' % (block.size, count))
RuntimeError: Incorrect number of samples (5505024 != 23104512), please report this error to MNE-Python developers
```

Aceticia, 2020-12-30:
Oh, never mind. I was loading using the wrong function.

drammock, 2020-12-30:
I tried this by downloading the data using `datalad` as recommended on the OpenNeuro website:

```console
$ datalad install https://github.com/OpenNeuroDatasets/ds002893.git
[INFO   ] Scanning for unlocked files (this may take some time)                                                                                   
[INFO   ] Remote origin not usable by git-annex; setting annex-ignore                                                                             
install(ok): /home/drmccloy/Desktop/ds002893 (dataset)
$ cd ds002893/sub-004/eeg/
$ ipython
```
followed by:
```
In [1]: import mne
In [2]: raw = mne.io.read_raw('sub-004_task-AuditoryVisualShift_run-02_eeg.set', preload=True)
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-2-00894410e2f8> in <module>
----> 1 raw = mne.io.read_raw('sub-004_task-AuditoryVisualShift_run-02_eeg.set', preload=True)

/opt/mne/python/mne/io/_read_raw.py in read_raw(fname, preload, verbose, **kwargs)
     80     ext = "".join(Path(fname).suffixes)
     81     if ext in readers:
---> 82         return readers[ext](fname, preload=preload, verbose=verbose, **kwargs)
     83     else:
     84         _read_unsupported(fname)

/opt/mne/python/mne/io/eeglab/eeglab.py in read_raw_eeglab(input_fname, eog, preload, uint16_codec, verbose)
    219     .. versionadded:: 0.11.0
    220     """
--> 221     return RawEEGLAB(input_fname=input_fname, preload=preload,
    222                      eog=eog, verbose=verbose, uint16_codec=uint16_codec)
    223 

<decorator-gen-222> in __init__(self, input_fname, eog, preload, uint16_codec, verbose)

/opt/mne/python/mne/io/eeglab/eeglab.py in __init__(self, input_fname, eog, preload, uint16_codec, verbose)
    317     def __init__(self, input_fname, eog=(),
    318                  preload=False, uint16_codec=None, verbose=None):  # noqa: D102
--> 319         eeg = _check_load_mat(input_fname, uint16_codec)
    320         if eeg.trials != 1:
    321             raise TypeError('The number of trials is %d. It must be 1 for raw'

/opt/mne/python/mne/io/eeglab/eeglab.py in _check_load_mat(fname, uint16_codec)
     57     """Check if the mat struct contains 'EEG'."""
     58     from ...externals.pymatreader import read_mat
---> 59     eeg = read_mat(fname, uint16_codec=uint16_codec)
     60     if 'ALLEEG' in eeg:
     61         raise NotImplementedError(

/opt/mne/python/mne/externals/pymatreader/pymatreader.py in read_mat(filename, variable_names, ignore_fields, uint16_codec)
     76 
     77     if not os.path.exists(filename):
---> 78         raise IOError('The file %s does not exist.' % (filename,))
     79 
     80     if ignore_fields is None:

OSError: The file sub-004_task-AuditoryVisualShift_run-02_eeg.set does not exist.
```

This is because the datafile is not really there, it's a (broken) link:
```console
$ ls -al sub-004_task-AuditoryVisualShift_run-02_eeg.set 
lrwxrwxrwx 1 drmccloy drmccloy 134 Dec 30 16:46 sub-004_task-AuditoryVisualShift_run-02_eeg.set -> ../../.git/annex/objects/0m/Wp/MD5E-s4434729--280e027e1863acb602f2e36312bd154c.set/MD5E-s4434729--280e027e1863acb602f2e36312bd154c.set
```
so after fetching the files:

```console
datalad get sub-004_task-AuditoryVisualShift_run-02_eeg.set
get(ok): sub-004/eeg/sub-004_task-AuditoryVisualShift_run-02_eeg.set (file) [from s3-PUBLIC...]
$ datalad get sub-004_task-AuditoryVisualShift_run-02_eeg.fdt 
[... 64 lines cut ...]
    416                 int(start_file), int(stop_file), cals, mult)

/opt/mne/python/mne/io/base.py in _read_segment_file(self, data, idx, fi, start, stop, cals, mult)
   1814 
   1815     def _read_segment_file(self, data, idx, fi, start, stop, cals, mult):
-> 1816         return self.__raw.__class__._read_segment_file(
   1817             self, data, idx, fi, start, stop, cals, mult)
   1818 

/opt/mne/python/mne/io/eeglab/eeglab.py in _read_segment_file(self, data, idx, fi, start, stop, cals, mult)
    364     def _read_segment_file(self, data, idx, fi, start, stop, cals, mult):
    365         """Read a chunk of raw data."""
--> 366         _read_segments_file(
    367             self, data, idx, fi, start, stop, cals, mult, dtype='<f4')
    368 

/opt/mne/python/mne/io/utils.py in _read_segments_file(raw, data, idx, fi, start, stop, cals, mult, dtype, n_channels, offset, trigger_ch)
    221             block = np.fromfile(fid, dtype, count)
    222             if block.size != count:
--> 223                 raise RuntimeError('Incorrect number of samples (%s != %s), '
    224                                    'please report this error to MNE-Python '
    225                                    'developers' % (block.size, count))

RuntimeError: Incorrect number of samples (5505024 != 23104512), please report this error to MNE-Python developers
```

drammock, 2020-12-30:
@Aceticia can you confirm that that particular file can be successfully read by MATLAB?  In other words, is this really an MNE-Python problem or is this particular file just corrupted?

cbrnr, 2020-12-31:
I can also confirm the problem with `sub-004_task-AuditoryVisualShift_run-02_eeg.set`. However, other files work (I picked three at random), so I'm guessing that this file is probably corrupted - but I don't have MATLAB/EEGLAB to confirm.

agramfort, 2021-01-04:
I confirm the file is broken. It fails to load with eeglab too.

closed 2021-01-04 by agramfort