mne-tools/mne-python #8681: New channel type: DBS

richardkoehler, 2020-12-28:
#### Describe the new feature or enhancement
I was wondering if it was possible to introduce a new channel type to MNE. Our group works mainly with LFP recordings obtained from DBS (deep brain stimulation) electrodes; I am aware that the channel type 'sEEG' exists, which comes kind of close, but it would be really advantageous if it were possible to specify the correct channel type. The channel type 'DBS' is also specified in iEEG-BIDS, so I was hoping that this might be worth thinking about - I didn't find an open issue on this, so I hope I didn't overlook anything.

#### Describe your proposed implementation
I'm more than happy to put in the work to make this possible, I am relatively new to python though, so I would need some help on what exactly needs to be done.

Thanks so much in advance and a happy holiday season to everyone :)

agramfort, 2021-01-04:
no objection on my side

the first thing to get accepted is a change in the fif format.

see 

https://github.com/mne-tools/mne-python/blob/master/mne/io/constants.py#L165
https://github.com/mne-tools/fiff-constants/blob/master/DictionaryTypes.txt#L245

@larsoner any suggestion for the channel type number to pick?

larsoner, 2021-01-04:
> The channel type 'DBS' is also specified in iEEG-BIDS, so I was hoping that this might be worth thinking about 

Indeed, I think we can just refer to https://www.nature.com/articles/s41597-019-0105-7 and state that this completes the iEEG-BIDS spec as we already have the other two ([ECoG and sEEG](https://github.com/mne-tools/fiff-constants/blob/master/DictionaryTypes.txt#L268-L270)). I'm not sure why we made ECoG 902, but sEEG is 802 and due to the similarity to sEEG it seems like 803 would be a good number for DBS. @richardkoehler can you make a PR to fiff-constants to add it to DictionaryTypes.txt? Then we can make a follow-up PR here to add it as a channel type.

richardkoehler, 2021-01-04:
Thanks so much for your answers! I made a PR, adding "dbs" as a new channel type with the number 803 to DictionaryTypes.txt. Hope I didn't overlook anything, but it seemed straightforward enough.

richardkoehler, 2021-01-07:
@larsoner , @agramfort 
Now that the pull request for DictionaryTypes was accepted, what else needs to be done? I already made a pull request to add it as a channel type to https://github.com/mne-tools/mne-python/blob/master/mne/io/constants.py#L165, but maybe this was premature.

larsoner, 2021-02-19:
I think we have this now from #8739

closed 2021-02-19 by larsoner