mne-tools/mne-python #8687: Some keyboard shortcuts are broken in raw.plot

cbrnr, 2020-12-30:
On macOS, toggling the scale bar (`s`) in an open `raw.plot` window doesn't work immediately. The scale bar is only toggled after e.g. switching to zen mode back and forth.

The help window shows incorrect shortcuts to increase/decrease window length and visible channels. For example, the window lists ⌘ + → as a shortcut to show a longer window, but the correct shortcut is fn + →. The same is true for the other three shortcuts.

F11 doesn't enter fullscreen mode, probably because this key is mapped to showing the desktop by default. I don't think we need a fullscreen shortcut on macOS because users can click the green button, but if we want to keep a shortcut we should change it to something else (at least on macOS where ^⌘F seems to be the default for full screen).

@drammock another one for you? If you don't have time I can also take a look (but not immediately).

drammock, 2020-12-30:
I'll fix it but I'll probably need your help testing the solution

drammock, 2020-12-30:
testing in a mac VM with the `macosx` backend, what I see is that pressing `s` will not immediately make a scalebar *disappear* if it's visible (but it will disappear as soon as some other event causes a redraw, such as window resize or zen mode toggle), but pressing `s` when the scalebars are hidden *will* work immediately.  Testing in the same VM with `qt5agg` backend, the `s` key works immediately as a toggle to both show and hide.  Is this consistent with what you're seeing @cbrnr ?

I also noticed that with the `macosx` backend that zen mode toggling shows/hides the scrollbars correctly, but doesn't actually resize the main axes to use the freed space when the scalebars are hidden.  This is true even if you trigger a draw event by resizing the window.

cbrnr, 2020-12-30:
Yes, this is consistent with what I see with the `macosx` backend. I didn't even know that everything should be enlarged in zen mode, but of course it does make sense.

PR #8688 (2020-12-30): MRG, VIZ: fix shortcuts and interactivity for raw.plot() with macosx backend
closes #8687 

- scalebar toggle now works correctly with `macosx` backend (previously it showed, but did not hide, scalebars immediately)
- zen mode now works correctly with `macosx` backend (previously it would show/hide the scrollbars but not use the freed space)
- help dialog corrected: macOS uses `fn+→` not `⌘+→` (similar for other 3 arrow keys)
- help dialog no longer shows `F11` as a fullscreen key for macOS (it didn't work, and fullscreen is built into the OS, so we just don't show a fullscreen key now)
closed 2020-12-30 by cbrnr