mne-tools/mne-python #8691: Causal spectral connectivity measures, e.g. DTF, (g)PDC

christianhacker, 2020-12-31:
#### Describe the new feature or enhancement
Additional (causal) spectral connectivity measures, such as partial directed coherence (Baccala & Sameshima). There are a couple of existing python packages [spectral_connectivity](https://github.com/Eden-Kramer-Lab/spectral_connectivity) and [connectivipy](https://github.com/dokato/connectivipy) that implement these. The latter relies on MVAR models, the former on multitaper. 

#### Describe your proposed implementation
Could be added as part of `mne.connectivity.spectral_connectivity` at least for multitaper mode. The reference implementation of (g)PDC requires fitting an MVAR model, which would necessitate adding another spectral estimation mode.

#### Describe possible alternatives
Aforementioned packages. Connectivipy (MVAR) could use some TLC; I think the author is done with it. 

#### Additional comments
None

cbrnr, 2020-12-31:
There is also https://github.com/scot-dev/scot which I have been involved with. I've been planning to integrate it into MNE (or at least adapt it so that it is compatible with MNE data structures), but I haven't really found the time. A comparison between these three packages would be useful before deciding which is easiest to integrate. Of course I'm biased towards SCoT, but on the other hand I could provide support if we decide to pick this package.

christianhacker, 2020-12-31:
I haven't been able to look at SCoT very much, plan to though. Does it have support for multitaper methods or just VAR models? The two packages I mentioned aren't exactly compliant with mne, but connectivipy does have some significance testing methods that could perhaps be integrated into `mne.stats`.

Another consideration would be adapting the relevant visualization modules (e.g. `mne.viz.plot_connectivity_circle`, `plot_sensors_connectivity`) to work with bidirectional connectivity measures. Shouldn't be too hard. 

What would be really neat would be some kind of source space visualization with connections between the dipoles. I don't know if this is possible in mne right now; don't understand how the 3d renderer works well enough. I've never used it, but the SIFT addon for EEGLAB appears to implement something [like that](https://sccn.ucsd.edu/wiki/Chapter_6.9._Visualization). It wouldn't have to be that intricate; I'd be happy with just color-coded tubes between dipoles.

closed 2022-08-26 by christianhacker