mne-tools/mne-python #8693: Several questions about source localization workflow

justinhyon, 2021-01-01:
Hello,
I am a student working with MNE and ran into several issues while working on source localization. I was wondering if anyone had any suggestions.

Firstly, I created my BEM surfaces using freesurfer's mri_watershed, and used these to try to make a source space. However, it appears as though the BEM surfaces are not scaled or located properly with respect to the brain, and the brain sticks out the back of the head. Here is plot_alignment with the BEM surfaces and the pial surface:
[image]
Here is the T1.mgz with the BEM inner skull in freeview:
[image]
This is not as far off as the pial surface, but the BEM is still quite clearly misaligned, particularly in the back of the head, and this leads to many sources being discarded since they are outside the inner skull.
I was wondering if anyone had any ideas on what is going wrong here? Is this something I would have to fix on the freesurfer side, or is there a quick fix i can implement in my MNE python code?

My second question is more simple: I want to get an average source estimate for each brain region in the atlas I use for my volume source space. I was wondering if there was a way to do this automatically, or if there was a way to output the number of sources that correspond to each label region, so that using these numbers and the order of labels in the atlas I could just average groups of rows in the final source estimate matrix.

Thank You!

agramfort, 2021-01-03:
there is clearly a pb

the white and pial surfaces are displayed correctly in freeview?

are you loading the correct surfaces?

can you try uwing the command line like:

mne freeview_bem_surfaces -s sample

where you replace sample by your subject name?

>

justinhyon, 2021-01-04:
The white and pial are also misaligned in freeview, and mne freeview_bem_surfaces seems to be further off than loading the bem and T1 separately into a blank freeview for some reason:
[image]
The surfaces are the ones that I generated using mri_watershed. They initially saved into the mri folder and were named incorrectly, so I changed them to the standard names (outer_skin.surf, etc) and moved them to the bem folder. 1 thing I noticed was that after the initial output from mri_watershed the names were in the format lh.name.surf_outer_skin_surface, but the surfaces had both hemispheres. The line I used was mri_watershed -surf braintest.surf T1.mgz braintest.mgz (-atlas also gave misaligned output)

agramfort, 2021-01-04:
can you share the script you used to produce these? did you use

https://mne.tools/dev/generated/commands.html#mne-watershed-bem

>

justinhyon, 2021-01-04:
When I tried mne watershed_bem it gave an error that it could not read the T1 volume, so I used the freesurfer function directly:
`mri_watershed -surf test.surf T1.mgz test.mgz`
adding -atlas still results in a shifted bem model.

The error that came up for mne watershed_bem is below. Should I change the parameters?
```
$mne watershed_bem -d . -s EM --overwrite

Running mri_watershed for BEM segmentation with the following parameters:

Results dir = ./EM/bem/watershed
Command = mri_watershed -useSRAS -surf ./EM/bem/watershed/EM ./EM/mri/T1.mgz ./EM/bem/watershed/ws

Running subprocess: mri_watershed -useSRAS -surf ./EM/bem/watershed/EM ./EM/mri/T1.mgz ./EM/bem/watershed/ws
error: mghRead(./EM/mri/T1.mgz, -1): could not open file

Mode:          use surfaceRAS to save surface vertex positions

Mode:          Saving out BEM surfaces

*********************************************************

The input file is ./EM/mri/T1.mgz

The output file is ./EM/bem/watershed/ws

mri_watershed Error: read failed

('\nMode:          use surfaceRAS to save surface vertex positions\nMode:          Saving out BEM surfaces\n\n*********************************************************\nThe input file is ./EM/mri/T1.mgz\nThe output file is ./EM/bem/watershed/ws\n\nmri_watershed Error: read failed\n\n', 'error: mghRead(./EM/mri/T1.mgz, -1): could not open file\x00\n')
```

agramfort, 2021-01-05:
can you provide an absolute path for subjects dir not just . ?

justinhyon, 2021-01-21:
I apologize as I was away for several weeks. Using the absolute path worked, thank you so much!

Does anyone have any ideas for the second issue? I will paste it again here:
I want to get an average source estimate for each brain region in the atlas I use for my volume source space. I was wondering if there was a way to do this automatically, or if there was a way to output the number of sources that correspond to each label region, so that using these numbers and the order of labels in the atlas I could just average groups of rows in the final source estimate matrix.
thank you!

agramfort, 2021-01-23:
@justinhyon this https://github.com/mne-tools/mne-python/pull/8777/files should make sure the pb does not occur again.

hoechenberger, 2021-01-23:
Hello @justinhyon 

> Does anyone have any ideas for the second issue? I will paste it again here:
> I want to get an average source estimate for each brain region in the atlas I use for my volume source space. I was [...]

This question seems to be a perfect fit for our new forum! Please post it there again – many more people will see it there, increasing your chances of getting a good response!

https://mne.discourse.group

closed 2021-01-24 by hoechenberger

larsoner, 2021-01-25:
> I want to get an average source estimate for each brain region in the atlas I use for my volume source space. I was wondering if there was a way to do this automatically, or if there was a way to [...]

[extract_label_time_course](https://mne.tools/stable/generated/mne.extract_label_time_course.html) should do this for you, see for example how it's used in https://mne.tools/dev/auto_tutorials/source-modeling/plot_visualize_stc.html#volume-source-estimates

PR #8777 (2021-01-24): enforce abs path
closes #8693