mne-tools/mne-python #8694: raw,plot

LF9076, 2021-01-03:
**READ THIS FIRST:** If you are having trouble getting MNE-Python to work with
your own data, you should ask for help on one of our other channels:

- [email list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis)
- [Gitter (chat)](https://gitter.im/mne-tools/mne-python)

Our GitHub issue tracker is only used to report bugs and suggest improvements
to MNE-Python. For any other questions, please use the email list or Gitter.
Usage questions that are posted as GitHub issues are usually closed without
being answered. See
[the FAQ entry on filing bug reports](https://mne.tools/dev/overview/faq.html#i-think-i-found-a-bug-what-do-i-do)
for more guidance. If you're pretty sure your problem is a bug in MNE-Python,
please **delete this section** and fill in the headings below, replacing the
placeholder text with your own responses. Surround any code samples with triple
backticks above and below the code block (see
[the GitHub guide to markdown](https://guides.github.com/features/mastering-markdown/#GitHub-flavored-markdown)
for help with issue formatting). Alternatively, you can put your MWE in a
[public gist](https://gist.github.com) and link to it in this issue.

#### Describe the bug
*Replace this text with a description of the bug.*

#### Steps to reproduce
*Replace this text with a code snippet or minimal working example [MWE] to
replicate your problem, using one of the [built-in datasets], preferably the
one called [sample]. If you can't replicate on a built-in dataset, provide also
a link to a small, anonymized portion of your data that does yield the error.*

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
*Replace this text with a description of what you expected to happen.*

#### Actual results
*Replace this text with the actual output, traceback, screenshot, or other
description of the results.*

#### Additional information
*Replace this text with the output of `mne.sys_info()`.*

cbrnr, 2021-01-03:
Probably something wrong with `data`, maybe the scaling? MNE expects data to be in Volts and not µV, so typical values are something like `1e-6`. Can you show some values (e.g. `data[:2, :5]`)?

closed 2021-01-03 by LF9076

msseibel, 2021-01-19:
Here, the obvious solution is to rescale the data array by something like `raw._data*=1e-6`.

I have a similar question but with MEG data. I dont wanted to scale the data directly, since I fear imprescision losses when scaling the data multiple times.
So I wanted to make use of the keys: `range`, `cal` and `unit_mul` . Therefore different scalings for `raw.info['chs'][i]['unit_mul'] `  were tested in the range 12, ... ,-12. But it has no affect on plotting. (On which functions does it have an effect?)

According to the Elekta Neuromag Functional Image File Format Documentation from March 2011, the relation between these three properties is: 
`raw._data*range*cal` and the unit of the result is defined by `unit` and `unit_mul`.  If `unit` is 112 (FIFF_UNIT_T) and `unit_mul` is -15 it means that data is provided in fT? And if `unit_mul` is 1 data should be interpreted as T?
What am I missing here?

So how can the setting of these properties be tested. I thought plotting would be the easiest way, or does the `scalings` in parameter `plot`  overwrites any effect of the three keys?

larsoner, 2021-01-19:
@msseibel most of the entries in `info` (except `'bads'`) are really meant to be read-only. Changing `info['chs'][...]` in particular is likely to cause problems down the line because we assume that MNE code is the only place that these get changed.

> I dont wanted to scale the data directly, since I fear imprescision losses when scaling the data multiple times.

In your case I would just scale with `raw.apply_function(lambda x: x * 1e-6)` or so. I wouldn't worry too much about losing precision from this -- do you think you will need to do this many times with many values during many save-load round-trips?

> Here, the obvious solution is to rescale the data array by something like raw._data*=1e-6.

Doing `raw._data*=1e-6.` is not a good idea because `_data` is a private attribute, and we could remove it or change it at any time without warning between releases so your code could break.

msseibel, 2021-01-19:
But when scaling the data directly is the correct way, what is then the meaning of the `unit_m`? 
Shouldn't `unit_m` also change when data is multiplied by a factor that represents a unit conversion?
E.g. when `x*=1e-3` then `unit_m` should be decreased by 3?

msseibel, 2021-01-19:
My actual problem is not directly related to the OPs problem, but I was unaware of how `mne` understands units and there scale when it comes to methods that are connected to the physical world e.g. `maxwell_filter` .  Now I see that it is actually explained in the notes of `mne.io.RawArray`. 
Thank you for your time.