mne-tools/mne-python #8699: varying bandwith for multitapers

behinger, 2021-01-05:
#### Describe the new feature or enhancement
I want to freely define the amount of temporal and spatial smoothing in a multitaper analysis. As far as I understand, currently, the bandwidth of the multitapers is a fixed value for all frequencies.
E.g. I want to have 1 Taper (thus no SNR increase) for frequencies up to 16Hz, with a frequency smoothing of 3/4 octave (thus changing time-windows), from 16Hz on, I want to fix the timwindow to 0.25s and use increasing amount of multitapers to smooth (and make gains from the SNR increase).

As an example this is how I would want to specify my TF analysis:
```python
freqs = np.arange(5,500,1)
 
#time-window= n_cycles / freq.
#freq_smoothing = time_bandwith/time-window

# The windows should have 3/4 octave smoothing in frequency domain
freq_window = freqs*2**(3/4/2) - freqs*2**(-3/4/2)

# The timewindow should be so, that for freqs below 16, it results in n=1
# Taper used, but for frequencies higher, it should be a constant 250ms.
time_window = 2/freq_window
time_window[freqs>=16] = 0.25

time_bandwidth = freq_window*time_window
n_cycles       = time_window*freqs
```

#### Describe your proposed implementation
I am not familiar enough with the current TF implementation to really help here.

#### Additional comments
Fieldtrip allows for such schemes and they are used in the literature (e.g. Hipp 2012)

larsoner, 2021-01-06:
> As an example this is how I would want to specify my TF analysis:

To put it differently, the short version is that you want `time_bandwidth` to be `float | ndarray, shape (n_freqs,)`? The `float` case being what we have in `master` and the `ndarray` case being what you would need for this?

> I am not familiar enough with the current TF implementation to really help here.

I am not 100% sure but I think this will require running `N` multitaper analysis where `N = len(np.unique(n_tapers))` and `n_tapers` is computed from `time_bandwidth`, `freqs`, and `n_cycles`, and hence lead to an `N`-fold slowdown. For your code snippet above with 500 frequencies this would require N=486 (yikes). It seems like something must be wrong with this reasoning because that would seemingly take ages, but it seems like changing the time_bandwidth changes the Slepian sequence used and hence you can't for example just sub-index from a single set of tapers in each case.

behinger, 2021-01-06:
Re short version: Yes exactly, similar to what the morlet implementation supports with n_cycles

and sorry, the frequency-vector was wrong, I usualy have logspaced 25 frequencies, half of them with more than 1 taper, so it is not that bad in practice (editted it).

larsoner, 2021-01-07:
@behinger do you want to give it a try to see how feasible it is? If you look and the code is too dense (or you do not have time) I can take a look

behinger, 2021-01-07:
Unfortunately, pretty busy with lockdown et. al. Generally I know that it is feasible, I have used it before (and others too, given it is implemented in fieldtrip, maybe worth looking into that implementation).
I am pretty busy at the moment, I even have another mne bugfix that I promised to upload and havent done yet. If you have time to look into it, that would be great!