mne-tools/mne-python #8700: Sorting out events from epoched data exported from EEGLAB

moonj94, 2021-01-05:
I am dealing with preprocessed and epoched data coming out of EEGLAB. There are two .set files to merge. However, when I attempt to concatenate_epochs, it says that the event and numbering don't align. This is because I need to input an events file. However, when I export an events file from EEGLAB, the format doesn't seem to be right. It keeps saying `can't convert string to float`.

So I created my own .txt file with the events formatted by one column of events (just the number). and then it gives me this error:

`ValueError: Unknown number of columns in event text file
None`

can anyone:
1) tell me what the correct format of the .eve or .txt file should be?
2) tell me if there is a better way to generate events file from eeglab? (I'm using brainvision and the vhdr or vmrk doesn't work)
3) on the side, how do I cut the existing epoch ( -1s : 2s) into (0:500ms)?

Thanks in advance.

agramfort, 2021-01-05:
can you share a file?

>

moonj94, 2021-01-05:
I have attached the EEGLAB file but removed the data itself. I have also attached the .txt file that I made.
[tmpEEG.mat.zip](https://github.com/mne-tools/mne-python/files/5772932/tmpEEG.mat.zip)
[S02_T01-eve.txt](https://github.com/mne-tools/mne-python/files/5772935/S02_T01-eve.txt)

agramfort, 2021-01-06:
please share a valid EEGLAB file .set with .fdt file

>

moonj94, 2021-01-06:
What are fdt files?

I uploaded the .set file. I had to change the number of trials, channels, and samples to 1 to make the size of the file small to upload.
[tmpEEG.set 2.zip](https://github.com/mne-tools/mne-python/files/5776403/tmpEEG.set.2.zip)

moonj94, 2021-01-06:
[tmpEEG.set 3.zip](https://github.com/mne-tools/mne-python/files/5776436/tmpEEG.set.3.zip)
This might be better

agramfort, 2021-01-06:
reading file with eeglab gives me:

pop_loadset(): loading file /Users/alex/Downloads/temp/tmpEEG.set ...
Warning: channel labels should not be empty, creating unique labels

and python:

ipdb> p ch_name
array([], dtype=float64)

so the pb is that you have no channel names

>

zxk11100, 2021-09-05:
Hi
Has this problem been solved? I have also encountered the same problem. My data is pre-processed, and it is segmented from -1s to 3S in EEGlab. Now I want to be segmented again, but when I used mne. epoch, it is not working and  the Error is  "The first argument to `Epochs` must be an instance of mne.io.BaseRaw". So how can I re-epoch this data?

agramfort, 2021-09-05:
@zxk11100 what you discribe here seems to be a different issue.

mne.Epochs are based on continues/raw data.

please ask this question on the mne discourse forum https://mne.discourse.group

cbrnr, 2021-09-15:
Closing due to inactivity of the OP.

closed 2021-09-15 by cbrnr