mne-tools/mne-python #8701: Add RMS plotting (merge_grad) function for plot_topomap.

AaronAngJW, 2021-01-06:
#### Describe the new feature or enhancement

Currently, merging of gradiometers (pair) reading is only possible with plot_topo but not in plot_topomap. This results in 'positive-only' outcome when using plot_topomap with 'grad' type sensors. 

#### Describe your proposed implementation
Simply enable to 'merge_grad' function in plot_topo to be available for plot_topomap as well.

#### Additional comments
Refer to thread https://mne.discourse.group/t/general-questions-gradiometer-readings-within-pair-produced-on-mne/2426 for more details.

agramfort, 2021-01-06:
what plot_topomap function are you taking about? the one in
evoked.plot_topomap does merge automatically.

please be more descriptive

>

AaronAngJW, 2021-01-06:
Hi Alex, apologies as I got mixed up with the technicality regarding how RMS works. Therefore, I am amending the initial issue drastically to address what I can ideally get. Please do refer to the edited text in the first post above.

dengemann, 2021-01-06:
This looks like a good idea. It could be nice to pull out the computation in plot_topomap to facilitate custom processing (I agree, currently "Ninja mode" is required for that).

agramfort, 2021-01-06:
here is a ugly (yet working hack)

```
import os
import numpy as np
import mne

data_path = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(data_path, 'MEG', 'sample',
                                    'sample_audvis_filt-0-40_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)

event_dict = {'auditory/left': 1, 'auditory/right': 2, 'visual/left': 3,
              'visual/right': 4, 'smiley': 5, 'buttonpress': 32}

events = mne.find_events(raw, stim_channel='STI 014')

epochs = mne.Epochs(raw, events, event_id=event_dict, tmin=-0.2, tmax=0.3)

evoked1 = epochs['auditory/left'].average().pick('grad')
evoked2 = epochs['auditory/right'].average().pick('grad')

def hack_evoked(evoked):
    data = evoked.data
    data = data.reshape((len(data) // 2, 2, -1))
    data = np.sqrt(np.mean(data ** 2, axis=1))
    evoked = evoked.copy().pick(np.arange(0, 2*len(data), 2))
    evoked.data = data
    return evoked

evoked1 = hack_evoked(evoked1)
evoked2 = hack_evoked(evoked2)

contrast = mne.combine_evoked([evoked1, evoked2], [1.0, -1.0])

# Drop bad channel

contrast.drop_channels(['MEG 2442'])

contrast.plot()
times = np.linspace(0, 0.2, 10)
contrast.plot_topomap(times=times, ch_type='planar1')
```

AaronAngJW, 2021-01-07:
Works like a charm on the sample audio data. On my side though, it collapses when I have only 203 gradiometer measurements in the data. I got a ValueError, which I am guessing, is the result of that? On the sample audio data, `data.shape = (204, 76)`. 

I assume 204 is the number of grads. When it is an odd number, `data.reshape((len(data) // 2, 2, -1))` fails.

agramfort, 2021-01-07:
yes you need to be smarter by dropping the pair of grads

>

AaronAngJW, 2021-01-08:
Thanks Alex! This was helpful.

AaronAngJW, 2021-01-20:
Hi again,

Sorry to keep this going but I just spotted something unusual.

[image]

The final contrast plot seems to be oscillating between the weights as shown. How do I go about remedying this?

agramfort, 2021-01-20:
sorry I don't get your concern. Maybe I don't see what you want me to see

>

AaronAngJW, 2021-01-20:
So for the top left arrows, the -0.100s and 0.000s contrast (middle plots) are likely to be driven by condition sponblinks/3 (eye blink) activity. But this is reflected as blue at -0.100s and red at 0.000s, when it is supposed to be red at both times.

agramfort, 2021-01-24:
@AaronAngJW you should compute the hack_evoked at the very end. For example interpolation of hack evoked will do terrible things. MEG interpolation is not like splines for EEG it's using the channels orientations for example.

AaronAngJW, 2021-01-26:
Oh great, that resolved the issue. Thanks Alex!

drammock, 2022-08-11:
closing: seems to be resolved, and > 1 year of inactivity.

closed 2022-08-11 by drammock