mne-tools/mne-python #8704: Notebook improvements

GuillaumeFavelier, 2021-01-07:
I open this issue following https://github.com/mne-tools/mne-python/pull/8503#issuecomment-748859979 to keep track of the missing features or anything that could be improved.

Features not supported :

- [ ] Google colab support? (asked in https://github.com/mne-tools/mne-python/pull/8503#issuecomment-751272941)
- [ ] Button/shortcut to take a screenshot
- [ ] Button/shortcut to save a movie
- [ ] Button/shortcut for autoplay
- [ ] Button/shortcut to display help
- [ ] Label time course mode

Design/UX:

- [ ] big square button (asked in https://github.com/mne-tools/mne-python/pull/8503#issuecomment-749411804)

My testing notebook is available on [gist.github.com](https://gist.github.com/GuillaumeFavelier/b470330fa71cfdef4c7be94b966ef49b).

GuillaumeFavelier, 2021-01-07:
About Google Colab, I made some research @agramfort.

Considering that [`ipyvtk_simple`](https://github.com/Kitware/ipyvtk-simple)'s main viewer class, [`ViewInteractiveWidget`](https://github.com/Kitware/ipyvtk-simple/blob/master/ipyvtk_simple/viewer.py#L30), is based on [`ipycanvas`](https://github.com/martinRenou/ipycanvas)'s [`Canvas`](https://github.com/martinRenou/ipycanvas/blob/master/ipycanvas/canvas.py#L245) widget, I found that [it is a known issue](https://github.com/martinRenou/ipycanvas/issues/170). The short answer is (and I quote [Martin Renou](https://github.com/martinRenou/ipycanvas/issues/170#issuecomment-755171874)):

> Custom widgets libraries are not supported by Google colab yet

We could track the progress on this matter on https://github.com/martinRenou/ipycanvas/issues/170 or more directly in https://github.com/googlecolab/colabtools/issues/498.

Anyway, I think those need to be resolved before more can be expected from `ipyvtk_simple` and finally `mne-python`.

GuillaumeFavelier, 2021-01-07:
Exactly what I was looking for, thank you @larsoner :+1:

GuillaumeFavelier, 2021-04-14:
The mouse events on Firefox might be fixed by https://github.com/Kitware/ipyvtklink/pull/25. I'll open a PR and give it a try.

GuillaumeFavelier, 2021-04-14:
It might not be necessary, I don't have those issues anymore on Firefox 87.0

GuillaumeFavelier, 2021-04-29:
For `Jupyter Lab` integration it might worth trying again once https://github.com/pyvista/pyvista/issues/1116 is fixed.

GuillaumeFavelier, 2022-03-09:
Quick update: `ipycanvas` now works on Google Colab. It's a big milestone for the `notebook` 3d backend.

https://github.com/martinRenou/ipycanvas/issues/170#issuecomment-1062071346

I'll check `ipyvtklink` and report back in here.

agramfort, 2022-03-09:
this would be a game changer !

Message ID: ***@***.***>
>

GuillaumeFavelier, 2022-03-09:
Although now `ipycanvas` is supported on Google Colab, `vtk`, even on notebook, still requires a running X server.

My testing workflow is: `ipycanvas` :heavy_check_mark: -> `ipyvtklink` :heavy_check_mark: -> `pyvista (ipyvtklink)` -> `mne (notebook)`

It's a work in progress but I managed get it working at low resolution, which is already amazing IMO.
Here is a modified version of [`ipyvtklink`'s default example](https://github.com/Kitware/ipyvtklink#python-vtk):

https://user-images.githubusercontent.com/18143289/157433174-fa8e5a99-83b6-46c6-89b3-aa1f116d92db.mp4

---

The important points:
1) setting the custom widget manager

```
from google.colab import output
output.enable_custom_widget_manager()
```

2) setting up `xvfb`

```
!sudo apt update
!sudo apt-get install libgl1-mesa-dev xvfb
!sudo apt install -yqq libxkbcommon-x11-0 libxcb-icccm4 libxcb-image0 libxcb-keysyms1 libxcb-randr0 libxcb-render-util0 libxcb-xinerama0 libxcb-xfixes0 libopengl0
```

3) start `xvfb` with the correct `DISPLAY` variable at the beginning of the cell

```
!/sbin/start-stop-daemon --start --quiet --background --exec /usr/bin/Xvfb -- :99 -screen 0 800x600x24 -ac +extension GLX +render -noreset

import os
os.environ['DISPLAY'] = ':99'
```

larsoner, 2023-10-01:
I think we can close this now that the notebook seems to be working correctly (even nowadays with `trame`)

closed 2023-10-01 by larsoner