mne-tools/mne-python #8705: Doc search on website is broken

hoechenberger, 2021-01-07:
The search doesn't work for me anymore in the dev documentation. Stable is still functional.

https://mne.tools/dev/search.html?q=compute_rank

doesn't produce any results and the JS console throws an exception, `ReferenceError: Can't find variable: Stemmer`

I'm using Safari 14.0.2

cc @drammock

drammock, 2021-01-13:
This is now true of the stable site as well.  Upstream reports [8623](https://github.com/sphinx-doc/sphinx/issues/8623) and [8603](https://github.com/sphinx-doc/sphinx/issues/8603) both say it's a theme problem, not a core sphinx problem.  I spent some time today trying to locally fix it by copying [the fix from the sphinx-rtd theme](https://github.com/readthedocs/sphinx_rtd_theme/pull/1021/files) and a couple other things, but it didn't work.  Might be time to prioritize #8184 ASAP?  Unless someone else claims it I'll take a look tomorrow to see just how painful it would be to transition to the pydata theme.

larsoner, 2021-01-13:
Did you try porting the sphinx_rtd_theme fix to sphinx_bootstrap_theme (what I assume we use under the hood)? I don't think #8184 will be very easy...

drammock, 2021-01-13:
Yes, that's what I was trying to say before. It's a one line change, and it didn't work.

agramfort, 2021-01-13:
these issues suggest it's fixed by latest versions. Am I missing something?

>

larsoner, 2021-01-13:
Latest versions of sphinx_rtd_theme, for people that use it -- we don't. So we need to figure out which extension / code is causing our problem, and fix it wherever that is.

larsoner, 2021-01-13:
> It's a one line change, and it didn't work.

I tried locally to make the change in `sphinx_bootstrap_theme` and it seemed to work -- maybe you changed a line in the wrong spot, or didn't `make clean` (or at least kill `search.html`) between tests or something? Will open a PR to test on CircleCI locally to confirm, upstream PR is https://github.com/ryan-roemer/sphinx-bootstrap-theme/pull/213

closed 2021-01-13 by drammock

drammock, 2021-01-13:
> maybe you changed a line in the wrong spot, or didn't `make clean`

I definitely did `make clean` and it's hard to believe that I pasted in the wrong spot... but who knows. It was late in the day and I was ready for dinner so many levels of stupidity are possible under those conditions 😂

PR #8742 (2021-01-13): MAINT: Fix search
Ideally we'd wait for the upstream PR to be merged and for a release to be made. But in the meantime we can fix the search by merging this, and update our `requirements_doc.txt` as those other steps occur.

That is assuming this actually fixes search. Current other PR builds like [this one](https://24504-1301584-gh.circle-artifacts.com/0/dev/index.html) show the issue so if this PR fixes it we should be good.

Closes #8705