mne-tools/mne-python #8706: in nicolet.py there is a problem reading files with non-round file length in seconds

miriamguen, 2021-01-07:
**READ THIS FIRST:** If you are having trouble getting MNE-Python to work with
your own data, you should ask for help on one of our other channels:

- [email list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis)
- [Gitter (chat)](https://gitter.im/mne-tools/mne-python)

Our GitHub issue tracker is only used to report bugs and suggest improvements
to MNE-Python. For any other questions, please use the email list or Gitter.
Usage questions that are posted as GitHub issues are usually closed without
being answered. See
[the FAQ entry on filing bug reports](https://mne.tools/dev/overview/faq.html#i-think-i-found-a-bug-what-do-i-do)
for more guidance. If you're pretty sure your problem is a bug in MNE-Python,
please **delete this section** and fill in the headings below, replacing the
placeholder text with your own responses. Surround any code samples with triple
backticks above and below the code block (see
[the GitHub guide to markdown](https://guides.github.com/features/mastering-markdown/#GitHub-flavored-markdown)
for help with issue formatting). Alternatively, you can put your MWE in a
[public gist](https://gist.github.com) and link to it in this issue.

#### Describe the bug
*Replace this text with a description of the bug.*

#### Steps to reproduce
*Replace this text with a code snippet or minimal working example [MWE] to
replicate your problem, using one of the [built-in datasets], preferably the
one called [sample]. If you can't replicate on a built-in dataset, provide also
a link to a small, anonymized portion of your data that does yield the error.*

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
*Replace this text with a description of what you expected to happen.*

#### Actual results
*Replace this text with the actual output, traceback, screenshot, or other
description of the results.*

#### Additional information
*Replace this text with the output of `mne.sys_info()`.*

agramfort, 2021-01-07:
@miriamguen can you share a problematic file?

miriamguen, 2021-01-07:
sure!
 35010100_0006.data
<https://drive.google.com/file/d/1EdhLV7WYhCOKO8Mv97MbSRCyaH9QWgR7/view?usp=drive_web>
 35010100_0006.head
<https://drive.google.com/file/d/1d0aETqnivycFwpW-EIT7cM3JgSbYOOPu/view?usp=drive_web>

-- 
-- 

Miriam Guendelman, MD

PHD student

Computational Psychiatry Lab <http://computational-psychiatry.com/>
Dept. of Brain and Cognitive Science
<http://in.bgu.ac.il/bcs/Pages/default.aspx>s
<http://in.bgu.ac.il/bcs/Pages/default.aspx>
Ben-Gurion University of the Negev
<http://in.bgu.ac.il/en/Pages/default.aspx>

Beer-Sheva, 84105

Israel
Cell: +972-50-2616202

agramfort, 2021-01-07:
@miriamguen see https://github.com/mne-tools/mne-python/pull/8712

PR #8712 (2021-01-08): fix int/float conversion in nicolet header
closes #8706

cc @miriamguen
closed 2021-01-08 by larsoner