mne-tools/mne-python #8718: set_bipolar_reference is too slow

jasmainak, 2021-01-08:
If you're doing bipolar reference on sEEG data with 180 contacts, then it is very slow. Here is an alternative script that does the same in a fraction of second:

```py
import numpy as np

import mne
from mne.datasets import sample

data_path = sample.data_path()

fname = data_path + '/MEG/sample/sample_audvis-ave.fif'

n_picks = 30  # change to bigger number to see difference in speed

# Reading evoked data
condition = 'Left Auditory'
evoked = mne.read_evokeds(fname, condition=condition, baseline=(None, 0),
                          proj=False)

del evoked.info['projs'][0]
evoked.pick_types(meg=False, eeg=True)
ch_names = evoked.info['ch_names']
evoked.pick_channels(ch_names[:n_picks])
ch_names = ch_names[:n_picks]

anode, cathode = ch_names[1:], ch_names[:-1]

n_channels, n_times = evoked.data.shape
n_virtual_channels = len(anode)
multiplier = np.zeros((n_virtual_channels, n_channels))
for idx, (a, c) in enumerate(zip(anode, cathode)):
    multiplier[idx, ch_names.index(a)] = -1. / n_times
    multiplier[idx, ch_names.index(c)] = 1

data_rereferenced = multiplier.dot(evoked.data)

mne.set_bipolar_reference(evoked, anode=anode, cathode=cathode)

data_rereferenced_master = evoked.data
assert np.allclose(data_rereferenced, data_rereferenced_master[:-1])

```

jasmainak, 2021-01-08:
Oh, it also seems like this function removes the channel locations. I'm not sure why because it means you can't do any processing downstream. I think the new channel locations should be the channel locations of the cathodes

marsipu, 2021-03-18:
I want to give it a shot.

PR #9270 (2021-04-22): ENH: Speed up set_bipolar_reference
## Reference issue
Fixes #8718

## What does this implement/fix?
This improves the performance of adding a larger number (>25) of bipolar channels as suggested by @jasmainak.
EDIT: A gist to measure the speed can be found [here](https://gist.github.com/marsipu/73f10b5303599837c8f182f0ab3f9e99)

## Additional information
I encountered some difficulties with how to best organize the creation of the new channels with the new method and handling Info while staying conservative. I think a large part of the performance improvement comes from moving `add_channels`  [... cut]
closed 2021-04-22 by larsoner