mne-tools/mne-python #8724: output of tfr_array_multitaper is incorrect?

chapochn, 2021-01-08:
#### Describe the bug
the output of tfr_array_multitaper seems incorrect

#### Steps to reproduce
```
import mne.time_frequency as tfr
import numpy as np
import matplotlib.pyplot as plt

signal = np.zeros(100)
signal[50:] = 1

tfr_out = tfr.tfr_array_multitaper([[signal]], sfreq=100, freqs=[20], time_bandwidth=2, output='power')[0, 0, 0]
plt.figure()
plt.plot(tfr_out)
```
#### Expected results
There shouldn't be any oscillations, the curve should be smooth, monotonically increasing then decreasing.

#### Actual results
There are oscillations, proportional to the number of cycles set as parameter in the function.

#### Additional information
Platform:      macOS-10.15.7-x86_64-i386-64bit
Python:        3.8.5 (default, Aug  5 2020, 03:39:04)  [Clang 10.0.0 ]
Executable:    /Users/.../opt/anaconda3/envs/mne/bin/python
CPU:           i386: 12 cores
Memory:        16.0 GB
mne:           0.22.0
numpy:         1.19.4 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.5.3
matplotlib:    3.3.1 {backend=MacOSX}
sklearn:       0.24.0
numba:         Not found
nibabel:       3.2.1
nilearn:       0.7.0
dipy:          1.3.0
cupy:          Not found
pandas:        1.1.5
mayavi:        4.7.2
pyvista:       0.27.4 {pyvistaqt=0.2.0, OpenGL 4.1 ATI-3.10.19 via AMD Radeon Pro 560X OpenGL Engine}
vtk:           9.0.1
PyQt5:         5.12.3

agramfort, 2021-01-09:
indeed there is something weird going on.

using fieldtrip I tried:

```
dat = zeros(1, 100);
dat(:, 51:end) = 1;
time = (1:100) / 100;
toi = time;
taper = 'dpss';
taper = 'hanning';
cyclenum = 7.;
foi = 20.
t_ftimwin = cyclenum ./ foi;

[spectrum_mtmconvol,ntaper,foi,toi] = ft_specest_mtmconvol(dat, time, 'timeoi', toi, 'timwin', t_ftimwin, 'taper', ...
taper, 'dimord', 'chan_time_freqtap');

plot(squeeze(abs(spectrum_mtmconvol(1,:,:))));
```

it gives me:

[image]

chapochn, 2021-01-15:
One should probably also add a test that would catch that

agramfort, 2021-01-16:
first we need to find the bug

I don't have much time unfortunately

>

chapochn, 2021-01-18:
I did some more tests, and actually the output seems correct. It occurs because the tapers don't go to 0 on the edges, so when there is discontinuity in the signal, there is also some discontinuity/oscillations in the output.
I think one enhancement that would be helpful, is to be able to specify the number of tapers and not only the time_bandwidth. Currently the number of tapers is imposed as floor(time_bandwidth - 1). However, the last tapers are always nonzero on the edges, so it could be good to choose a smaller number of tapers to have no discontinuity/oscillations. It should be pretty easy to add that option. I could look into it.

agramfort, 2021-01-24:
but if you set time_bandwidth to 2 than you have only 1 taper and it does not look at all like fieldtrip.

can you give a code snippet that produces similar results as fieldtrip?

chapochn, 2021-01-27:
Hi,
This is my fieldtrip code:

```
dat = zeros(1, 100);
dat(:, 51:end) = 1.;
time = (1:100) / 100;
toi = time;
taper = 'dpss';
cyclenum = 7.;
foi = 20.;
t_ftimwin = cyclenum ./ foi;
BW = 1.  % time x half-bandwidth, corresponds to time_bandwidth=2 in mne
tapsmofrq = BW/t_ftimwin

[spectrum_mtmconvol,ntaper,foi,toi] = ft_specest_mtmconvol(dat, time, ...
    'tapsmofrq', tapsmofrq, 'timwin', t_ftimwin, 'freqoi', [foi], 'taper', taper)

plot(squeeze(abs(spectrum_mtmconvol).^2));
```

It gives the same result as mne (module some scaling and cyclic convolution).

chapochn, 2021-11-21:
So I think this is not an issue. The only potential enhancement would be to specify manually the number of tapers instead of it being set automatically. Although it would be nice, I don't personally need it right now, so I'll close this issue.

closed 2021-11-21 by chapochn