mne-tools/mne-python #8729: Adding the option to measure connectivity over time to spectral_connectivity()

Avoide, 2021-01-11:
#### Describe the new feature or enhancement
Hi all,

#7937 indicated there was an error in the PLI implementation and I also had the same problem when working with resting-state epoched data. But after looking more into it I found out it is not really an error but instead two different ways to analyze connectivity: over time or over trials.

Mike X Cohen explains the difference quite nicely in his video here:
[Youtube Link](https://youtu.be/2h2X4TVUF5I?t=554)

I also commented in that issue, but are now creating a standalone feature request.
The current PLI and PLV implementation in spectral_connectivity() is connectivity over trials, which are suited for ERP data. But for resting-state data it does not make sense to analyze connectivity over trials but should instead be over time.

#### Describe your proposed implementation
I tried implementing my own version to calculate PLV and wPLI over time and you can see a simple working example here:

```
# Libraries
import numpy as np
import mne
from scipy import special
from mne.connectivity import spectral_connectivity

# Generate data
np.random.seed(42)
n_epochs = 5
n_channels = 3
n_times = 1000
data = np.random.rand(n_epochs, n_channels, n_times)

# Set sampling freq
sfreq = 250 # A reasonable random choice

# Choose what kind of data should be used for evaluation
# 0 = same trial repeated in all epochs
# 1 = 10Hz sinus waves with random phase differences in each channel and epoch
data_option = 1

if data_option == 0:
    # Make all 5 epochs the same trial to show difference between connectivity
    # over time and over trials. Here we expect con over trials = 1
    for i in range(n_epochs):
        data[i] = data[0]
elif data_option == 1:
    # Generate 10Hz sinus waves to show difference between connectivity
    # over time and over trials. Here we expect con over time = 1
    for i in range(n_epochs):
        for c in range(n_channels):
            wave_freq = 10
            epoch_len = n_times/sfreq
            # Introduce random phase for each channel
            phase = np.random.rand(1)*10
            # Generate sinus wave
            x = np.linspace(-wave_freq*epoch_len*np.pi+phase,
                            wave_freq*epoch_len*np.pi+phase,n_times)
            data[i,c] = np.squeeze(np.sin(x))
else:
    print("Data_option value chosen is invalid")

# Define freq bands
Freq_Bands = {"delta": [1.25, 4.0],
              "theta": [4.0, 8.0],
              "alpha": [8.0, 13.0],
              "beta": [13.0, 30.0],
              "gamma": [30.0, 49.0]}
n_freq_bands = len(Freq_Bands)
# Convert to tuples for the mne function
fmin=tuple([list(Freq_Bands.values())[f][0] for f in range(len(Freq_Bands))])
fmax=tuple([list(Freq_Bands.values())[f][1] for f in range(len(Freq_Bands))])

# Connectivity methods
connectivity_methods = ["plv","wpli"]
[... 136 lines cut ...]
    con_data1[1,epoch,:,:,:] = PLV_WPLI_con[1] # weighted phase lag index

# Take average across epochs for PLV and wPLI
con_data2 = np.mean(con_data1,axis=1)
# Save to final array
con_data_time[0,:,:,:] = con_data2[0] # phase locking value
con_data_time[1,:,:,:] = con_data2[1] # weighted phase lag index

print("Alpha PLV over time")
print(con_data_time[0,:,:,2])
```

Notice I added the option to choose what kind of data should be used. If data_option = 0 then the expected output is that PLV across trials are 1 since I just repeated the first epoch in all other epochs, while if data_option = 1 the PLV across time should be 1 as I use 10Hz sine waves with different phase differences.

It is still quite rough and not optimized but should give an idea of the procedure.
I am not completely certain about my implementation but the wPLI is calculated by following equation 1 in [1] and weighing by the absolute magnitude of the phase differences as described by [2]. PLV was calculated according to the equation in [3]. According to #8305 this equation should be equivalent to what is being used in spectral_connectivity() already, but my implementation over time gives another PLV than spectral_connectivity().

It would be helpful if anyone could have a look. After optimizing it the feature should probably just be added as an option to spectral_connectivity() and does not need its own version.

#### Describe possible alternatives
I don't know any alternatives as this suggestion is just about a slight different way of calculating the connectivity over time instead of over trials.

[1] Hardmeier, Martin, Florian Hatz, Habib Bousleiman, Christian Schindler, Cornelis Jan Stam, and Peter Fuhr. 2014. “Reproducibility of Functional Connectivity and Graph Measures Based on the Phase Lag Index (PLI) and Weighted Phase Lag Index (WPLI) Derived from High Resolution EEG.” PLoS ONE 9 (10). https://doi.org/10.1371/journal.pone.0108648.
[2] Vinck, Martin, Robert Oostenveld, Marijn Van Wingerden, Franscesco Battaglia, and Cyriel M.A. Pennartz. 2011. “An Improved Index of Phase-Synchronization for Electrophysiological Data in the Presence of Volume-Conduction, Noise and Sample-Size Bias.” NeuroImage 55 (4): 1548–65. https://doi.org/10.1016/j.neuroimage.2011.01.055.
[3] Lachaux, Jean Philippe, Eugenio Rodriguez, Jacques Martinerie, and Francisco J. Varela. 1999. “Measuring Phase Synchrony in Brain Signals.” Human Brain Mapping 8 (4): 194–208. https://doi.org/10.1002/(SICI)1097-0193(1999)8:4<194::AID-HBM4>3.0.CO;2-C.

mmagnuski, 2021-01-15:
@Avoide I updated your code example to inclue syntax coloring. I'll try to add to the discussion on monday!

balandongiv, 2021-07-07:
Hi @larsoner , May I know whether this issue and suggestion has been incorporate in recent mne version?

mmagnuski, 2021-07-07:
No, but the connectivity measures are moved to a separate repository now - mne-connectivity. It would be good to discuss this issue there once the transition is over