mne-tools/mne-python #8733: ValueError in mne.viz.plot_evoked_joint

BarryLiu97, 2021-01-12:
I am using SEEG data, and I want to plot evoke joint figure, while I've got this problem.
```
    mne.viz.plot_evoked_joint(evoke)
  File "D:\SEEG_Cognition\SEEG\lib\site-packages\mne\viz\evoked.py", line 1458, in plot_evoked_joint
    colorbar=False, **topomap_args_pass)
  File "D:\SEEG_Cognition\SEEG\lib\site-packages\mne\evoked.py", line 356, in plot_topomap
    nrows=nrows, ncols=ncols)
  File "D:\SEEG_Cognition\SEEG\lib\site-packages\mne\viz\topomap.py", line 1540, in plot_evoked_topomap
    _prepare_topomap_plot(evoked, ch_type, sphere=sphere)
  File "D:\SEEG_Cognition\SEEG\lib\site-packages\mne\viz\topomap.py", line 103, in _prepare_topomap_plot
    exclude='bads')
  File "D:\SEEG_Cognition\SEEG\lib\site-packages\mne\io\pick.py", line 446, in pick_types
    _check_meg_type(meg)
  File "D:\SEEG_Cognition\SEEG\lib\site-packages\mne\io\pick.py", line 348, in _check_meg_type
    % (allowed_types, meg))
ValueError: meg value must be one of ['grad', 'mag', 'planar1', 'planar2'] or bool, not seeg
```
my code is pretty simpe, just mne.viz.plot_evoked_joint(evoke)

BarryLiu97, 2021-01-12:
Here is the data's link.
```
https://drive.google.com/file/d/1zIinPymoWgHlKav594EHyekU8zjewVS-/view?usp=sharing
```

drammock, 2021-01-13:
@BarryLiu97 FYI you often don't need to manually close issues when there's an open PR that address them. There are 2 reasons:

1. Someone might realize a problem with the open PR and it might not get merged, in which case the issue is not fixed in the master branch and should remain open
2. Github will automatically close this issue when the PR does get merged, because I linked them together (notice how the PR description says "closes # 8733", that is what makes them linked)

BarryLiu97, 2021-01-13:
> @BarryLiu97 FYI you often don't need to manually close issues when there's an open PR that address them. There are 2 reasons:
>
> 1. Someone might realize a problem with the open PR and it might not [...]

Got it .Thank u for the heads up! I always thought I should close the issues if the developers have figured out why

drammock, 2021-01-13:
> I always thought I should close the issues if the developers have figured out why

The preferred approach might be different for different software projects.  Here we try to always use the "linking PR with issue" approach so that they get automatically closed when the PR is merged.  In this repo, one case where you still *should* close manually is if it becomes clear that the problem was not with MNE-Python (i.e., if it turned out to be a corrupt data file, or an error in your script, etc).

BarryLiu97, 2021-01-14:
Again, thanks.

PR #8736 (2021-01-14): MRG, VIZ, FIX: sEEG picking in _prepare_topomap_plot()
closes #8733 

Not sure if this is a reasonable fix since I've never personally worked with sEEG, but based on general knowledge I'm assuming there's nothing special needed as far as combining sensor pairs?  Can someone confirm?

cc @BarryLiu97.  On this branch using your code from #8733 I get this figure:

[image]
closed 2021-01-14 by larsoner