mne-tools/mne-python #8737: Add volume support in the point spread/cross talk functions

mdclarke, 2021-01-12:
**Describe the problem**
MNE currently does not support volume source estimates in the point spread and cross talk functions.

**Describe your solution**
Add support for volume source estimates and volume labels.

larsoner, 2021-01-12:
This should hopefully be doable hopefully without too much work by refactoring the label extraction code -- it should have the mechanics for going from volumetric atlas to activations. @olafhauk do you want to take a look?

Also @olafhauk does the PSF/CTF code properly support free ori inverses? It seems like at least [`make_inverse_resolution_matrix`](https://mne.tools/dev/generated/mne.minimum_norm.make_inverse_resolution_matrix.html#mne.minimum_norm.make_inverse_resolution_matrix) does.

dasdiptyajit, 2021-05-03:
@larsoner hi, any idea why the current  'make_inverse_resolution_matrix'  function doesn't include a normal inverse orientation?As I can see the orientation is limited to i.e., pick_ori = None/vector.

larsoner, 2021-05-04:
@dasdiptyajit if you make a fixed orientation inverse I think it should work. Are you talking about using a loose inverse but with `pick_ori='normal'`? We could maybe add support for this. WDYT @olafhauk ?

dasdiptyajit, 2021-05-04:
@larsoner yes, you are right, I was talking about inverse with a loose constraint and one more thing, the PSFs are likely not to conserve their signs when you compute the resolution matrix with loose inverse, at-least what I have seen so far. What could be the reason for that? thanks!

pettetmw, 2022-01-28:
This is related... `SourceSpaces` created by `mne.setup_volume_src_space()` using default parameter `volume_label=None`, will create a single, whole-head volume.  A `Forward` operator created with this will trigger an uncaught exception when passed to `get_point_spread()`:

```
#   File "resolution_matrix.py", line 102, in _get_psf_ctf
#     vertno_rh = src[1]['vertno']
#   IndexError: list index out of range
```

`_get_psf_ctf()` apparently expects the forward operator to be constructed from two `SourceSpaces`, as is the convention for surface source estimates.  While it is possible to work around this by dividing the whole-head volume into left and right `SourceSpaces`, we still face the issue described here #10187.

agramfort, 2022-01-28:
Yes this needs a fix to deal with more than historical cortical source
spaces.

> Message ID: ***@***.***>
>

larsoner, 2022-01-28:
@olafhauk in the case of free-orientation inverses, is the right (or at least one not totally wrong) thing to do here to compute cross-talk/point-spread values for each of the 3 directions independently? Then one could decide how to combine those three values (max, mean, l2-norm, etc.)?

PR #11083 (2022-08-31): ENH: Add support for vol to PSF/CTF
Closes #8737

Marking as draft so that @olafhauk can add a tutorial, perhaps :)
closed 2022-08-31 by agramfort