mne-tools/mne-python #8740: Enhancements to modified Beer Lambert Law

rob-luke, 2021-01-12:
#### Describe the new feature or enhancement
In the discussion at https://github.com/mne-tools/mne-python/pull/8711 the following enhancements for the BLL were mentioned. I am listing them here so these suggestions don't get lost.

- [ ] Add support for alternative extinction coefficients.
- [ ] Refactor code to have private function where source-detector distance can be specified.
- [ ] Add some references to literature.
- [ ] Add age dependent partial path length factor ([reference](https://www.spiedigitallibrary.org/journals/Journal-of-Biomedical-Optics/volume-18/issue-10/105004/General-equation-for-the-differential-pathlength-factor-of-the-frontal/10.1117/1.JBO.18.10.105004.full?SSO=1))

#### Additional comments
The code works fine as is. This is just optional improvements for when anyone has time. Please list additional feature requests below too.

AdmerStroh, 2024-02-06:
Hi! I would like to suggest a new feature to this function. It should be possible to use a ppf per wavelength, similar to what is possible in Homer3 where ppf is a vector of factors per wavelength.

see https://github.com/BUNPC/Homer3/blob/master/FuncRegistry/UserFunctions/hmrR_OD2Conc.m

larsoner, 2024-02-08:
@AdmerStroh would you be up for adding support for this to MNE-Python?