mne-tools/mne-python #8743: ValueError  in   mne.viz.plot_sensors_connectivit

BarryLiu97, 2021-01-13:
Hi, I was trying to learn https://mne.tools/stable/auto_examples/connectivity/plot_sensor_connectivity.html#sphx-glr-auto-examples-connectivity-plot-sensor-connectivity-py.I change the method to 'coh', aka the coherence, and I've got this error:
```
ValueError: zero-size array to reduction operation maximum which has no identity
```
And here is my code
```
import mne
from mne import io
from mne.connectivity import spectral_connectivity
from mne.datasets import sample
from mne.viz import plot_sensors_connectivity

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'

raw = io.read_raw_fif(raw_fname)
events = mne.read_events(event_fname)

raw.info['bads'] += ['MEG 2443']

picks = mne.pick_types(raw.info, meg='grad', eeg=False, stim=False, eog=True,
                       exclude='bads')

event_id, tmin, tmax = 3, -0.2, 1.5  # need a long enough epoch for 5 cycles
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0), reject=dict(grad=4000e-13, eog=150e-6))
fmin, fmax = 3., 9.
sfreq = raw.info['sfreq']  # the sampling frequency
tmin = 0.0  # exclude the baseline period
epochs.load_data().pick_types(meg='grad')  # just keep MEG and no EOG now
con, freqs, times, n_epochs, n_tapers = spectral_connectivity(
    epochs, method='coh', mode='multitaper', sfreq=sfreq, fmin=fmin, fmax=fmax,
    faverage=True, tmin=tmin, mt_adaptive=False, n_jobs=1)

plot_sensors_connectivity(epochs.info, con[:, :, 0])
```

BarryLiu97, 2021-01-13:
I think I may find the error, I debug the code and I find this:

[image]

mmagnuski, 2021-01-15:
I'm not sure if this is related to your problem, but I remember finding recently that `spectral_connectivity` ignores negative tmin or tmax (so you can't for example compute pre-stimulus connectivity).

BarryLiu97, 2021-01-16:
> I'm not sure if this is related to your problem, but I remember finding recently that `spectral_connectivity` ignores negative tmin or tmax (so you can't for example compute pre-stimulus connectivity).

I just find something odd. My data(epoch)'s tmin is -0.5 and tmax is 2.0  I use -0.5 ~ 0 as the baseline
[image]
And I set the ```tmin=0``` in spectral_connectivity
[image]
However, the length of the  connectivity I got is 2501(for the sampling rate is 1000Hz), it seems that it see the -0.5 time point as the 0 time point.

chazuttu, 2021-01-16:
This type of error comes up when the trace is empty