mne-tools/mne-python #8745: ica.get_sources seems to be broken

jasmainak, 2021-01-14:
here is the script to reproduce:

```py
import mne
from mne.preprocessing import ICA
from mne.datasets import sample

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'

raw = mne.io.read_raw_fif(raw_fname)
raw.pick_types(meg=False, eeg=True, exclude='bads', stim=True).load_data()

ica = ICA(n_components=0.95, method='fastica').fit(raw)

ica_sources = ica.get_sources(raw, start=20, stop=40)
data, times = ica_sources[:]

ica_sources.plot()
assert data.shape == times.shape
assert abs(times[0] - 20) < 1
```

I get:

```py
ValueError: shape mismatch: objects cannot be broadcast to a single shape
```

mmagnuski, 2021-01-15:
I can reproduce. BTW the `start` and `stop` args work differently depending on the type of the value: samples if int, seconds if float. I'm not opposed to logic like this, but I is easy to get wrong.

cbrnr, 2021-01-15:
> samples if int, seconds if float.

😱 - what? Any chance we could change that? I thought we'd always uses seconds...

larsoner, 2021-01-15:
Actually for `raw` most places IIRC the convention is typically to support only `int` in samples, e.g.

https://mne.tools/dev/generated/mne.io.Raw.html#mne.io.Raw.get_data

So I'd rather go the other way. But either way I agree it's probably more dangerous than it is useful to support both. One way out of this mess is to deprecate `start` and `stop` in favor of clearer names like `t_start` and `t_stop` or `tmin` and `tmax` which would be floats. Or have `start` and `stop` be in samples to be consistent with `get_data` and `save` etc. but add `t_start` and `t_stop` which would be in seconds...

larsoner, 2021-01-15:
... actually `save` uses `tmin/tmax`

https://mne.tools/dev/generated/mne.io.Raw.html#mne.io.Raw.save

cbrnr, 2021-01-15:
Good suggestion! `tmin` and `tmax` should always refer to seconds, and `start`/`stop` should always refer to samples.

jasmainak, 2021-01-16:
humm ... now that I think of it, why does one even need `start`/`stop` since you can almost always use python indexing in place of it? E.g., I could have done:

```py
ica_sources = ica.get_sources(raw[20:40][0])
```

so you can almost get rid of these parameters

chazuttu, 2021-01-16:
This error implies that one of the variables being used  has a shape incompatible with another on the same line

mmagnuski, 2021-01-16:
I'm also +1 for `tmin` and `tmax` in seconds and `start`/`stop` in samples.

Thanks @chazuttu, we noticed that, but that means that there is a bug in the `get_sources()` method because `len(times)` and `data.shape[1]` should have the same length.

larsoner, 2021-01-28:
This appears to be fixed by #8801

closed 2021-02-05 by larsoner