mne-tools/mne-python #8750: New CTF acquisition software (v6.2.0) format incompatible with mne.io.read_raw_ctf

jstout211, 2021-01-14:
#### Describe the bug
We recently installed the new CTF acquisition software installed (v6.2.0) and mne.io.read_raw_ctf throws error on newly acquired data.  This is a repeatable error on newly acquired datasets.  Previous datasets acquired before the software update, continue to load properly.  

#### Steps to reproduce
Unable to attach dataset (too big) - here is a google drive link to the dataset: https://drive.google.com/file/d/1LRN7rmzjg-CkvrBh0Q-0SCnqizZHmT7X/view?usp=sharing

To reproduce:  raw = mne.io.read_raw_ctf(dataset)

#### Expected results
This still works for older datasets acquired before the acquisition upgrade.
[image]

#### Actual results
[image]

#### Additional information
In [4]: mne.sys_info()
********************************************************************************
WARNING: Imported VTK version (8.1) does not match the one used
         to build the TVTK classes (9.0). This may cause problems.
         Please rebuild TVTK.
********************************************************************************

Platform:      Linux-4.18.0-240.10.1.el8_3.x86_64-x86_64-with-redhat-8.3-Ootpa
Python:        3.7.8 | packaged by conda-forge | (default, Jul 31 2020, 02:25:08)  [GCC 7.5.0]
Executable:    /home/stoutjd/miniconda3/envs/mne/bin/python
CPU:           x86_64: 12 cores
Memory:        15.3 GB

mne:           0.21.1
numpy:         1.19.4 {blas=openblas, lapack=openblas}
scipy:         1.4.1
matplotlib:    3.3.3 {backend=Qt5Agg}

sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.2.0
cupy:          Not found
pandas:        1.1.4
dipy:          1.3.0
mayavi:        4.7.2
pyvista:       0.27.2 {OpenGL 4.6 (Core Profile) Mesa 20.1.4 via Mesa Intel(R) UHD Graphics (CML GT2)}
vtk:           8.1.2
PyQt5:         5.12.3

jstout211, 2021-01-14:
I am not sure if this is a suitable solution, but if I use the mne.io.read_raw_ctf(......., **system_clock='ignore**'), the dataset is able to be read into a proper mne variable. 

[image]

larsoner, 2021-01-14:
Glad that there is a workaround so that it's not a total blocker for people. It would be good to fix the system clock parsing for newer files, though. We'll have to figure out what the new way to deal with it is...

agramfort, 2021-01-14:
I can replicate the issue but I don't know the format well enough to see
how to fix it.

can you find maybe a matlab reader that works on these files?

>

jstout211, 2021-01-14:
Our lab contacted CTF about the issue.  They are going to take a look at the Channel 0 writeout.  It seems like the acquisition is currently just writing zeros - which MNE takes to mean the scan has been stopped (unless you ignore).  This may be a non-issue if they are able to correct this functionality.  And I don't suspect many labs have the acquisition software update so far (released in late Dec?).   I'll update the post once we hear back from CTF.

larsoner, 2021-04-19:
@jstout211 any reply from CTF about this?

jstout211, 2021-04-29:
I just pinged CTF about this for an update.  I will let you know what they say.

jstout211, 2021-04-29:
@larsoner  -  I was told that the clock channel was fixed in release:
ctfmeg-6.2.0-11844.el6_10.x86_64.rpm

larsoner, 2021-04-29:
Okay so it sounds like no action is needed at our end since the `ignore` is a sensible workaround for the affected files, and it should be fixed. I'll go ahead and close but feel free to reopen if newly acquired data is not correct @jstout211 -- and thanks for investigating with CTF!

closed 2021-04-29 by larsoner