mne-tools/mne-python #8751: MNE sEEG example

jasmainak, 2021-01-15:
The MNE sEEG example has many issues [which I first brought up here](https://github.com/mne-tools/mne-python/issues/8748#issuecomment-760574396)

- bad channels are not marked
- no referencing has been done? Or is the raw data already bipolar referenced? In any case, this info is missing from the tutorial
- some of the shafts are in the "air"
- the `stc.plot_3d` function gives the incorrect idea that there are lots of active regions in the brain. See comment above for alternative viz and also #8382 as point out by @larsoner 
- it's too verbose about transformations. I think it's just an identity transformation usually. 
- information about what task etc is missing. I think we shouldn't be plotting "something" just for the sake of plotting. There should be some neuroscience motivation at least :)

cc @adam2392 perhaps you have time to improve this? :)

agramfort, 2021-01-15:
yes we need a better dataset. The one we have is a fake made historically by @choldgraf 

where can we find such data?

maybe we can convert the data from https://neuroimage.usc.edu/brainstorm/Tutorials/Epileptogenicity in iEEG-BIDS and use this?

choldgraf, 2021-01-15:
[image]

adam2392, 2021-01-15:
Yes unfortunately I just grabbed some random SEEG electrode coordinates I could find and there was perhaps a z shift somewhere in that dataset... let me see if I can find a better set of coords.

agramfort, 2021-01-15:
@choldgraf I would love to have access to your folder of such GIFs :) you'll always impress me by their relevance to the context ;)

jasmainak, 2021-01-15:
I don't think GIFs suit you @agramfort ;)

btw, we can also look at openneuro. I even see a dataset uploaded by @adam2392 . But yes, something from Brainstorm is also good as we know what exactly to expect

choldgraf, 2021-01-15:
[image]

agramfort, 2021-01-15:
@adam2392 <https://github.com/adam2392> thoughts?

@choldgraf LOL !

>

adam2392, 2021-01-19:
Does mne have a way to pull datasets from openneuro?

If so, I can in the future utilize a dataset say that I uploaded to openneuro. Although the dataset I uploaded is all ECoG...
For the sEEG example, I'll try finding a better suited example that is inside the brain and localized correctly. No guarantees.

> maybe we can convert the data from https://neuroimage.usc.edu/brainstorm/Tutorials/Epileptogenicity in iEEG-BIDS and use this?

Where can we store the converted dataset?

drammock, 2021-01-19:
> Where can we store the converted dataset?

presumably on osf.io (that's what we use for other brainstorm datasets)

adam2392, 2021-01-19:
> > Where can we store the converted dataset?
> 
> presumably on osf.io (that's what we use for other brainstorm datasets)

Do we have a wiki/dev-documentation for how to get this setup? Once I have some time, I'm assuming I have to do something along these lines:

1. use mne-bids to convert Brainstorm dataset
2. upload to osf.io
3. setup link in mne-python to this dataset
4. substitute it in the seeg example
5. use mne-python data fetcher(?) to get the dataset during unit tests / example-rendering?

> Does mne have a way to pull datasets from openneuro?
> 
> If so, I can in the future utilize a dataset say that I uploaded to openneuro. Although the dataset I uploaded is all ECoG...
> For the sEEG example, I'll try finding a better suited example that is inside the brain and localized correctly. 

In the future, I can also look into adding the dataset I uploaded as an example related to epilepsy

jasmainak, 2021-01-19:
for sEEG I can probably help. See private communication @agramfort . Timeline is 8-12 months though :)

drammock, 2021-01-19:
@adam2392 edits to your sequence:

- step 2 requires permissions on our OSF account, so either we can grant you that or one of the other devs can upload it.
- Step 5 is slightly more complicated in that it the dataset fetching code is under revision (see #8679) but basically yeah, it involves adapting a `data_path()` function to work with `bst_epileptogenicity`.
- Steps 3 & 5 should happen at the same time (same PR).
- Step 4 would come after the combined step 3+5 (i.e., could be a subsequent PR) although presumably you would want to do the code changes for it early on to make sure the converted dataset looks the way you expect it to, before bothering to do the upload & other steps.

larsoner, 2023-10-01:
In the last couple of years @alexrockhill and others have worked on https://mne.tools/dev/auto_tutorials/clinical/20_seeg.html which hopefully takes care of a lot of these issues, but PR(s) welcome to fix any remaining ones!

closed 2023-10-01 by larsoner