mne-tools/mne-python #8755: Read ICA EEGLab: problems with AMICA decomposition

michaelcgold, 2021-01-15:
### (possible) Bug

`read_ica_eeglab()` appears to not be able to read in AMICA (note: *not* `runica`) decompositions from an EEGLab .set file.

The error upon trying to instantiate ICA:

```
  AssertionError                            Traceback (most recent call last)
 <ipython-input-10-a6f5a4e0501a> in <module>
       5 from mne import preprocessing
       6 
 ----> 7 ica = preprocessing.read_ica_eeglab('full_filepath')
       8 
       9 
 
 ~/opt/anaconda3/envs/mne/lib/python3.8/site-packages/mne/preprocessing/ica.py in read_ica_eeglab(fname)
    2754 
    2755     n_ch = len(ica.ch_names)
 -> 2756     assert eeg.icaweights.shape == (n_components, n_ch)
    2757     # When PCA reduction is used in EEGLAB, runica returns
    2758     # weights= weights*sphere*eigenvectors(:,1:ncomps)';
 
 AssertionError: 
```

### Steps to reproduce
 ICA was performed with EEGLab's [AMICA](https://github.com/japalmer29/amica/wiki/AMICA) (not runica) with PCA reduction to rank correct for interpolated channels.

Looking at the data in EEGLab, `ICAweights = n_components x n_pcs`, and `ICAsphere = n_pcs x n_channels`. I’m guessing `read_ica_eeglab` is assuming the `icaweights` matrix is already multiplied by `icasphere`.

### Expected results
`read_ica_eeglab` should correctly read in pca reduced IC decompositions from AMICA (and possibly other algorithms -- this might want to be tested, such as FASTICA).

### Minimal reproducible example
[61 channel recording (.set/.fdt)](https://github.com/mne-tools/mne-python/files/5822735/MNE_TestData.zip)

cbrnr, 2021-01-15:
@christian-oreilly do you have time to take a look?

christian-oreilly, 2021-01-15:
@cbrnr I cannot for the next week because I have coming deadlines, but I'll have a look at it after that.

michaelcgold, 2021-01-15:
Thanks! Also should note that IC rejection already occurred in this dataset (and all the datasets I’m planning to use).

mmagnuski, 2021-01-16:
I don't think AMICA is supported in mne right now. For example AMICA allows for multiple ICA models and provides a timecourse for their likelihood. Some of the ICs can be even shared between these models. Because of these differences mne-python could be able to import AMICA model only in the simplest scenario. But you are right - there might be an inconsistency in how ICA is read from eeglab in general, expecting no PCA reduction.

michaelcgold, 2021-01-16:
Hi, to clarify - the request is not to have functionality for reading in AMICA direct output/models. The expectation (and my problem) is that the appropriate models and time courses were selected using the postAmicaUtility in EEGLAB and single icaweights and icasphere matrices are in the .set file. Perhaps this the simplest scenario you had in mind?

mmagnuski, 2021-01-16:
@michaelcgold Yes, that should be the simplest scenario. :) In other words, your problem does not seem specific to AMICA (at least at the surface).

agramfort, 2021-01-24:
@michaelcgold see #8780

closed 2021-01-27 by cbrnr

michaelcgold, 2021-01-28:
Thanks everyone for your help!! Appreciate your work

PR #8780 (2021-01-27): MRG, BUG: fix IO or read_ica_eeglab
attempt to close #8755

my debug script using this branch.

```
from scipy import linalg
import matplotlib.pyplot as plt

import mne
from mne.viz.topomap import _find_topomap_coords

ica = mne.preprocessing.read_ica_eeglab('ICAamica_MNE_test.set')
# ica.info = raw.info
# ica._update_ica_names()
# ica.plot_components()

unmixing = ica.icaweights @ ica.icasphere
mixing = linalg.pinv(unmixing)

pos = _find_topomap_coords(ica.info, picks=None)
plt.close('all')
mne.viz.plot_topomap(mixing[:, 7], pos=pos)
plt.show()
```

here are the ICA topos I get using EEGLAB

 [... cut]