mne-tools/mne-python #8756: trouble installing mne on anaconda python 3.8.5

moonj94, 2021-01-16:
I have a fresh install of anaconda and it seems to be python version 3.8.5. I have been stuck at this state  for a while now:

```
(base) Jaes-MacBook-Pro:~ jaemoon$ conda install -c conda-forge mne
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: | 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
Examining conflict for matplotlib scikit-image mne matplotlib-base nltk jupyte- 

```

Does mne work with python 3.8.5? and what can I do to make this  work?

drammock, 2021-01-16:
It does work with python 3.8.5.  I see `nltk` in that conflicts list, and it seems like some items in the conflicts list are not being shown...  would it work for you to install `mne` into a separate environment instead of `base`?  That is our [recommended setup](https://mne.tools/stable/install/mne_python.html#installing-mne-python-and-its-dependencies) and will ensure that the mne environment only contains packages that we know to be compatible.

chazuttu, 2021-01-16:
pip install -U mne

hoechenberger, 2021-02-13:
Closing due to inactivity, feel free to re-open if needed.

closed 2021-02-13 by hoechenberger

MSA8D8, 2021-06-27:
Hello everyone. I want to report that I was also faced with the same issues. 
But, I was able to avoid this issues by using pip command to install mne. 
So, please feel free to read this error report and remove this in the future update. 

Here is what happened to me.
----
I re-install the anaconda3 due to other problems, so I once uninstall anaconda3 completely. 
However, after installing the latest one (anaconda3-2021.05), I was faced with this issue with the error message below.
The version of Python is 3.8.10. 

```
% conda install -c conda-forge mne
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: | 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
Examining conflict for sortedcollections sphinx pytz pycodestyle three-merge / r
```

hoechenberger, 2021-06-27:
Hello, you should install MNE into its own, new environment, as described in our official installation instructions to avoid such issues.