mne-tools/mne-python #8761: misc issues with plotting in MNE

jasmainak, 2021-01-18:
Grouping a bunch of misc viz issues that I faced when using MNE:

1. If you do:

```py
epochs.info['bads'] = [epochs.ch_names[0]]
epochs.plot()
```

it does not show the channel name in gray. Instead it doesn't plot it. This is not technically a bug but inconsistent with what `raw.plot()` does. I think it would be useful to see the bad channel also in this plot.

2. It would be nice to have a scale also in the `epochs.plot` as in `raw.plot()`. It really helps improve the interpretability of the data.
3. The `scale` shown in `raw.plot()` is incorrect if `scalings='auto'` since the scale is channel-wise. The scalings/ylim issue seems to be common across multiple viz functions
4. In `raw.plot_psd` the usual matplotlib controls to zoom in, "home button" etc are missing. Not sure if it's only my set up. Happy to share more details if folks can't reproduce easily.
5. Selecting a channel in `evoked.plot()` should make the line plot bold so we can see that channel more clearly. I think this is how it used be before. Not sure if there were any reasons to change it?

drammock, 2021-04-21:
> The `scale` shown in `raw.plot()` is incorrect if `scalings='auto'` since the scale is channel-wise. The scalings/ylim issue seems to be common across multiple viz functions

if `scalings='auto'` then scaling is *not* channel-wise, it is still done jointly by channel type.  I don't think the scalebars are actually wrong; try 

```py
fig1 = raw.plot()
fig2 = raw.plot(scalings='auto')
```
and then adjust one of them using `+` and `-` keys until the scalebars for a given channel type are showing approx the same number; the signals should look approx the same.

drammock, 2021-04-21:
> Selecting a channel in `evoked.plot()` should make the line plot bold so we can see that channel more clearly. I think this is how it used be before. Not sure if there were any reasons to change it?

I don't recall it ever doing that, and that part of the code [hasn't changed significantly in 5-6 years](https://github.com/mne-tools/mne-python/blame/main/mne/viz/evoked.py#L43-L67).  Feel free to open as a separate feature request if you want.

larsoner, 2021-04-21:
>  Instead it doesn't plot it. This is not technically a bug but inconsistent with what raw.plot() does. I think it would be useful to see the bad channel also in this plot.

For this (point 1 above) you can do `picks='all'` for example. The documented default `None` selects "good data channels", which is why you don't see your bad channel. I think it ends up being plotted in red, though (?). I think this difference is okay since if you've chosen picks explicitly you probably want to clearly see the bad channel (and it would be harder to see in gray), whereas in raw plotting the space is always allocated to the bad channel (and it's included by default).

I'll close since it seems like all other items are crossed off now!

closed 2021-04-21 by larsoner