mne-tools/mne-python #8765: EEG electrode coordinates: commonly used alias mappings

sappelhoff, 2021-01-20:
coming from https://github.com/mne-tools/mne-python/pull/8753#issuecomment-762463945 

When plotting EEG data *without* having measured (digitized) all electrode positions, users fall back on template positions (in 3D or 2D). 
Some electrode positions have multiple names (aliases), depending on from which convention they are named.
For example, The T3 and T4 electrode positions are synonymous with the T7 and T8 electrode positions.

In this thread we want to collect such aliases (with sources that support our argument) (i) for documentation purposes, and (ii) to make some kind of feature or API available to users that they receive an appropriate electrode position from a template - even when passing an alias of an electrode position that is not present in the electrode position file itself.

---

The following is what I posted in https://github.com/mne-tools/mne-python/pull/8753#issuecomment-762698328 ... let's continue the discussion here:

sure, I think that'd be an important feature. Can we start a list on synonymously used names?

- Cbz, Cb1, Cb2 (apparently also CBz, CB1, CB2 with capital B)
    - stands for "cerebellar", in the 5% article, synonymous to Iz, I1, I2
    - needs to be verified, because the Chatrian paper has a slightly different definition ... and @cbrnr and I thought that they would be POO7 and POO8
    -  There is also a bunch of other CB* electrodes, see figs below

- T3/T4 are synonymous with T7/T8 (see: https://core.ac.uk/download/pdf/144965483.pdf)

- T5/T6 are synonymous with T9/T10 (see: https://core.ac.uk/download/pdf/144965483.pdf)

- A1/A2 seem to be placed **on** the ears? (see figure in https://core.ac.uk/download/pdf/144965483.pdf)
    - does somebody have more precise information on that?
    - if they are placed **on** the ears, these are different from LPA and RPA (for info, see: https://www.fieldtriptoolbox.org/faq/how_should_i_report_the_positions_of_the_fiducial_points_on_the_head/)

Also there is much more, looking at these two figures from

> Chatrian GE, Lettich E, Nelson PL. Ten percent electrode system for topo-graphic studies of spontaneous and evoked EEG activity. Am J EEGTechnol 1985;25:83±92.

[image]

- M1/M2 is probably TP9/TP10 ?
    - standing for "mastoid" ... and TP9 / TP10 are typically pretty much on the mastoid

- T1/T2 are probably another synonym for T9/T10 ??? (like T5/T6 ???)

- Chatrian also includes ocular electrode positions:
    - So1 and So2 (superior occipital)
    - Io1 and Io2 (inferior ...)
    - Lo1 and Lo2 (lateral ...)
    - --> these are different by subject (pretty much like A1 and A2), because everyone's eyes (and ears) are slightly different and difficult to measure consistently ... or (for example) you might end up with an electrode too near or too far from the outer canthus of an eye

tczhangzhi, 2021-01-20:
Thanks for all your information, it saves my day. It seems most aliases are included, so I would like to talk about the implementation. In my opinion, there are several potential types of API we can utilize to set aliases, I am wondering which type is the best (or something better).
### 1. WHAT TO SET
I believe the best way to solve this problem is to create an `alias template` whose content may be:
```
name, alias
Cb1, POO7
Cb2, POO8
```
because we do not want to couple the alias with every single montage template and modify them every time when we find new aliases.
During utilization, users can decide whether to utilize it for mapping by passing `'auto'|'none'|dict` to `alias`, `'auto'` to utilize our provided aliases, `'none'` to not utilize, or pass a `dict` to use custom aliases.
### 2. WHERE TO SET
The position to provide the `alias` parameter is a big problem and need a trade-off.
2.1 set aliases for `ch_names` and map them to names included in the montage template:
```
info = mne.create_info(ch_names=ch_names, ch_types=ch_types, sfreq=sampling_freq, aliases='auto')
```
In this way, the user can pass the `'auto'` to utilize our provided aliases, `'none'` to not utilize, or pass a `dict` to use custom aliases. Then, while calling `set_montage`, we can translate the `ch_names` to the corresponding standard names included in montage template. But it is a little wired that we set the aliases in `info` but we cannot translate them at `mne.io.RawArray` stage, because we do not know the template the user uses.
```
raw = mne.io.RawArray(data, info)
montage = mne.channels.make_standard_montage('some_standard')
raw.set_montage(montage)
```
2.2 set aliases for montages and map the channel names in a montage to user definition.
```
info = mne.create_info(ch_names=ch_names, ch_types=ch_types, sfreq=sampling_freq)
raw = mne.io.RawArray(data, info)
montage = mne.channels.make_standard_montage('some_standard', aliases='auto')
raw.set_montage(montage)
```
Similarly, we do not know the channel names defined by users, so we cannot directly translate standard channel names to their aliases, but we can store them and plot both if the aliases are set.
2.3 keep both `ch_names` and channel names in montages unchanged, only care about aliases while mapping. It is not flexible (because the user can not plot montages with aliases and can only plot the standard one) but very clean.
```
info = mne.create_info(ch_names=ch_names, ch_types=ch_types, sfreq=sampling_freq)
raw = mne.io.RawArray(data, info)
montage = mne.channels.make_standard_montage('some_standard')
raw.set_montage(montage, aliases='auto')
```
Personally, I like the third one, but I am interested in your opinions because your suggestions are very useful during the PR.

mmagnuski, 2021-01-20:
@tczhangzhi I like the idea to set aliases during `.set_montage()` step.

sappelhoff, 2021-01-22:
I also like `2.3` with doing alias mapping during `.set_montage()`

> it is not flexible (because the user can not plot montages with aliases and can only plot the standard one) but very clean.

but wouldn't you be able to do:

```python
raw.set_montage(montage, aliases='auto')

# now the montage associated with "raw" has the aliases
raw.plot_sensors()
```

and then see the montage with aliases? see [plot_sensors](https://mne.tools/stable/generated/mne.io.Raw.html?highlight=mne%20io%20raw#mne.io.Raw.plot_sensors)

it'd be slightly more impractical (you need a "RAW" and set the montage before you can plot it) .. but not impossible.

One more step, next to the API and the "alias.csv" file you are suggesting, I recommend that we document our decisions somewhere ... why is X an alias of Y --> put a short string somewhere, potentially with a DOI reference. this could go into a third column of "alias.csv" WDYT?

tczhangzhi, 2021-01-24:
I like the idea of plotting sensors **with aliases** using `raw.plot_sensors()` instead of `montage.plot(kind='topomap')`. The montage template should follow the standard and users can custom the visualization of `raw`, i.e., the input data. It makes sense.

> One more step, next to the API and the "alias.csv" file you are suggesting, I recommend that we document our decisions somewhere ... why is X an alias of Y --> put a short string somewhere, potentially with a DOI reference. this could go into a third column of "alias.csv" WDYT?

I totally agree with this suggestion. And one more thing, how should we provide the information of existing aliases and their description for the ones who want to know the details? Or should we consider it to be transparent to users, and we can guide users to open `alias.csv` in the official document for details of existing aliases?

tczhangzhi, 2021-01-28:
Seems no one is working on it, so I following the above discussion and provide an implementation.

PR #8799 (2021-02-10): Add `match_alias` option for `set_montage()`
#### Reference issue
Example: Fixes #8765.

#### What does this implement/fix?
Add `match_alias` option for `set_montage()` following the discussion of [PR8735](https://github.com/mne-tools/mne-python/pull/8753) and [ISSUE8765](https://github.com/mne-tools/mne-python/issues/8765).

#### Additional information
Following the existing API style, we change the original idea of providing an option `alias='auto'|'none'|dict`. Instead of that, we follow the style of option `match_case` which is utilized to handle the case conversion problem and provide a s [... cut]
closed 2021-02-10 by drammock