mne-tools/mne-python #8766: IndexError: list index out of range    in   mne.io.read_epochs_eeglab

BarryLiu97, 2021-01-20:
Here is my coe:
```
import mne
fpath = 'fear'
epoch = mne.io.read_epochs_eeglab(fpath)
```
Then an error occurred:
```
  File "<input>", line 1, in <module>
  File "D:\SEEG_Cognition\SEEG\lib\site-packages\mne\io\eeglab\eeglab.py", line 279, in read_epochs_eeglab
    uint16_codec=uint16_codec)
  File "<decorator-gen-223>", line 24, in __init__
  File "D:\SEEG_Cognition\SEEG\lib\site-packages\mne\io\eeglab\eeglab.py", line 471, in __init__
    event_latencies.append(events[ev_idx].latency)
IndexError: list index out of range
```
Here is my data https://drive.google.com/file/d/1F7NczB4gjqdcX-1wf705Dm7Z8BXYmdV0/view?usp=sharing

closed 2021-01-20 by BarryLiu97

guoyongjian111, 2021-11-24:
How did you solve it? I met the same problem，thanks

BarryLiu97, 2021-11-25:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    My problem comes from wrong operation in eeglab
    
        
            
        
    
    
     
    
        
                    
            
                
                    
                            
                                
                            
                            
                                3273376560
                            
                    
                        
                            
                                    ***@***.***
                            
                        
                
            
        
        
    
    
        签名由
        网易邮箱大师
        定制